GrumpyOldPizza/ArduinoCore-stm32l0 #2: Cant see USB port on a MAC on 1.8.5

trlafleur, 2018-03-18:
I have been unable to get IDE 1.8.5 to see the USB port to allow programing of this device on a mac...

Apple's USB prober see's the device, but its never available in the IDE...

Any ideas???

~~~
Full Speed device @ 49 (0x14213000): .............................................   Composite device: "STM32  BOOTLOADER"
    Port Information:   0x0018
           Not Captive
           External Device
           Connected
           Enabled
    Number Of Endpoints (includes EP0):   
        Total Endpoints for Configuration 1 (current):   1
    Device Descriptor   
        Descriptor Version Number:   0x0100
        Device Class:   0   (Composite)
        Device Subclass:   0
        Device Protocol:   0
        Device MaxPacketSize:   64
        Device VendorID/ProductID:   0x0483/0xDF11   (STMicroelectronics)
        Device Version Number:   0x2200
        Number of Configurations:   1
        Manufacturer String:   1 "STMicroelectronics"
        Product String:   2 "STM32  BOOTLOADER"
        Serial Number String:   3 "04B23643006B"
    Configuration Descriptor (current config)   
        Length (and contents):   45
        Number of Interfaces:   1
        Configuration Value:   1
        Attributes:   0xC0 (self-powered)
        MaxPower:   100 ma
        Interface #0 - Application Specific/Device Firmware Update ..............................................   "@Internal Flash  /0x08000000/1536*128g"
        Interface #0 - Application Specific/Device Firmware Update (#1) ..............................................   "@Option Bytes  /0x1FF80000/01*032 e"
        Interface #0 - Application Specific/Device Firmware Update (#2) ..............................................   "@DATA Memory /0x08080000/2*3Ke"

~~~

GrumpyOldPizza, 2018-03-20:
What happens if you simply upload the first sketch ? Like "Blink" ? At that point it should had over control to dfu-utils, which should see the "STM32  BOOTLOADER". Before you have the first sketch loaded there will be no serial port available (assuming that's what you tried to do).

GrumpyOldPizza, 2018-03-21:
> Not really sure what your asking here.... I’m stuck at a point where my Mac
> Arduino IDE can’t see the serial port - USB port. So not sure how to load
> any program. As noted OSX sees the port...
>
> I have tried to run a program, but I donot see the correct port in list of
> available ports. Maybe we have a chicken and egg problem?

What happens if you just try to download a sketch ? What does the IDE emit
as error message ?

trlafleur, 2018-03-21:
> auth/AEXCK7Zt1eTyE1GPZvYO_

I can see the USB device in the IDE..

I can compile a demo program , not issue
But when I select Tools-->port the device  is NOT listed there, so one
CAN'T start as download as the USDB device is NOT visible....

as noted in prior email, OSX see the device, but its NOT available to the
IDE....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

GrumpyOldPizza, 2018-03-21:
> But when I select Tools-->port the device is NOT listed there, so one
> CAN'T start as download as the USDB device is NOT visible....

That is kind of the key point ... "Tools -> port" just looks for the USBD
serial device, which is actually not used to download the sketch. If
present it's used to reboot the board into bootloader mode, which then pops
up as "STM32  BOOTLOADER" in the USB device list.

On my system (just verified again to make sure nothing changed lately), if
you simply download the sketch and no port is open, it will just start the
"upload" utility.

This is why I asked "can you simply push the download button on the IDE and
report any error messages from there".

- Thomas

trlafleur, 2018-03-21:
Will try and see what happens.   This is not the usual action one takes on most boards.  

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

> But when I select Tools-->port the device is NOT listed there, so one
> CAN'T start as download as the USDB device is NOT visible....

That is kind of the key point ... "Tools -> port" just looks for the USBD
serial device, which is actually not used to download the sketch. If
present it's used to reboot the board into bootloader mode, which then pops
up as "STM32 BOOTLOADER" in the USB device list.

On my system (just verified again to make sure nothing changed lately), if
you simply download the sketch and no port is open, it will just start the
"upload" utility.

This is why I asked "can you simply push the download button on the IDE and
report any error messages from there".

- Thomas

wrote:

trlafleur, 2018-03-21:
Suffix successfully added to file
Using library LoRaWAN at version 1.0.0 in folder:
/Users/lafleur/Desktop/MySensor-Arduino-development/Arduino_fork/hardware/TleraCorp/stm32l0/libraries/LoRaWAN

Using library STM32L0 at version 1.0 in folder:
/Users/lafleur/Desktop/MySensor-Arduino-development/Arduino_fork/hardware/TleraCorp/stm32l0/libraries/STM32L0

Sketch uses 72512 bytes (36%) of program storage space. Maximum is 196608
bytes.
Global variables use 6192 bytes (30%) of dynamic memory, leaving 14288
bytes for local variables. Maximum is 20480 bytes.
Forcing reset using 1200bps open/close on port /dev/cu.SLAB_USBtoUART6
/Users/lafleur/Desktop/MySensor-Arduino-development/Arduino_fork/hardware/TleraCorp/stm32l0/tools/macosx/stm32l0-upload
0x1209 0x6665
/var/folders/53/f8mlg0xn1xv3fsm58phqmtb40000gn/T/arduino_build_731149/LoRaWAN_OTAA_US915.ino.dfu

10
9
8
7
6
5
4
3
2
1
FAIL

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

GrumpyOldPizza, 2018-03-22:
Can you hold the BOOT button while you toggle RESET on the board. Of course
USB should be connectected all the time.

Then run form a command shell:

/Users/lafleur/Desktop/MySensor-Arduino-development/Arduino_fork/hardware/TleraCorp/stm32l0/tools/macosx/dfu-util
-l

I do not quite understand the output that you send me. It implies that
there is a USB port to toggle (which you said was not there), and then it
fails to flash, while it's already in bootloader mode (as per previous
e-mail). So let's see whether the flash utility throws an error, or reports
something unexpected.

- Thomas

GrumpyOldPizza, 2018-03-27:
No feedback for 5 days. Verified that dfu-utils properly works on a Mac (thanx Simon).

closed 2018-03-27 by GrumpyOldPizza