GrumpyOldPizza/ArduinoCore-stm32l0 #4: STM32L0.stop and timers

ssozonoff, 2018-04-04:
Hi,

Not exactly clear on how device comes out of stop() mode. Would you kindly explain what is going on in this small attached code sample.

Why am I seeing the following print statements ? 

_Serial enabled!
Hello World
Going to sleep
Do callback Loratx
Do callback Loratx
Do callback Loratx
Do callback Loratx
Do callback Loratx
Do callback Loratx_

I would have expected to see the loop execute until the following sleep... and I can only assume the reason this code works at all is that the timer callback also wakes up the device ?

Kind thanks,
Serge

**Code Sample**

```
#include <Arduino.h>
#include <STM32L0.h>
#include "TimerMillis.h"

TimerMillis LoRaTimer;

void setup() {
    
  Serial.begin(115200);
  delay(4000);
  Serial.println("Serial enabled!");
 
  LoRaTimer.start(callbackLoRaTx,  5000, 35000);  // 5 minute period, 1 minute delay
}

void loop() {

    Serial.println("Hello World");
    delay(500);
    
    Serial.println("Going to sleep");
    STM32L0.stop();        // Enter STOP mode and wait for an interrupt
    delay(1000);
}

void callbackLoRaTx(void)
{           
    Serial.println("Do callback Loratx");    
}
```

GrumpyOldPizza, 2018-04-05:
I do need to clarify and fix this.

As of right now, STM32L0.stop() and STM32L0.sleep() wake only up if you
call STM32L0.wakeup().

Some user observable change will also make it wake up (like receiving data
from a UART). Need to reduce the set of automatic wakeup, to make this less
confusing.

- Thomas

ssozonoff, 2018-04-06:
OK however in .stop() the interrupt still fires which in turn calls .wakeup() if I understand correctly ?

Thanks

GrumpyOldPizza, 2018-04-06:
Correct.

closed 2018-04-25 by ssozonoff