GrumpyOldPizza/ArduinoCore-stm32l0 #5: LoRaWAN.disableChannel() not working?

DavidBoelee, 2018-04-17:
Hi,

I am trying to setup a single channel node, so that I can combine it with a single channel gateway. I am using the Add B-L072Z-LRWAN1 in combination with your (great!) library.

I found LoRaWAN.disableChannel() by snooping into the source coude and figured I could use this to, well, disable channels. So that I end up with a single channel gateway. So my setup now looks like this:

void setup( void )
{
    LoRaWAN.begin(EU868);

for (unsigned int x = 1; x <=15; x++)
 {  
    LoRaWAN.disableChannel(x);
 }
   // LoRaWAN.setADR(false);
    LoRaWAN.setDataRate(0);
    LoRaWAN.setTxPower(10);
    LoRaWAN.setSubBand(2); // for TTN
    LoRaWAN.joinOTAA(appEui, appKey, devEui);
}

But... There are still transmissions on other channels than 0 as well, as you can see on the screenshot below.

[image]

Am I using disableChannel() wrong, or could it be something else is wrong?

Thanks,

David

GrumpyOldPizza, 2018-04-17:
Semi-bug.

You set the initial mask. However the JOIN_ACCEPT reply from the gateway
will most likely add channel 4/5/6/7/8.

So you need to put the channel disable post the joinOTAA().

DavidBoelee, 2018-04-17:
Ah! That makes sense. Not even a semi-bug, but just how the protocol works.
Thanks for your quich answer.

GrumpyOldPizza, 2018-04-17:
> 4287-11e8-8f59-db99e72851a9.png
> fguYSINIsprOMocks5tpkWSgaJpZM4TY7eE

Yeah, too a bit to find out. There is also this issue of enabling/disabling
default channels. I had thought that it might enable the default channels
behind you back, post OTAA join.

DavidBoelee, 2018-04-18:
It does a little bit more than that. I experimented with it today and found out that you have to disable the channels  before every package you send. If you don't, other channels are activated again. Also, it's a good idea to disable channels before joining, so that no time is wasted sending join messages on channels that aren't listened to by the single channel gateway. It may feel a little bit sketchy to keep disabling channels, but hey, single channel is a bit sketchy anyways. ;)

Code below worked reliable for me.

void setup( void )
{
    LoRaWAN.begin(EU868);

    for (unsigned int x = 1; x <=15; x++)
 {  
    LoRaWAN.disableChannel(x);
 }

   // LoRaWAN.setADR(false);
    LoRaWAN.setDataRate(0);
    LoRaWAN.setTxPower(10);
    LoRaWAN.setSubBand(2); // for TTN

    LoRaWAN.joinOTAA(appEui, appKey, devEui);

}

void loop( void )
{
    delay(10000);

    if (!LoRaWAN.busy() && LoRaWAN.joined())
    {
        for (unsigned int x = 1; x <=15; x++)
        {  
        LoRaWAN.disableChannel(x);
        }
        LoRaWAN.beginPacket();
        LoRaWAN.write(0xef);
        LoRaWAN.write(0xbe);
        LoRaWAN.write(0xad);
        LoRaWAN.write(0xde);
        LoRaWAN.endPacket();
    }
}

GrumpyOldPizza, 2018-04-18:
Need to trace that throu. Something does not feel right.

There are a bunch of problems with the original LoRaMac-node code. One is
the default channels, which you can modify, are not properly restored. So
even if you disabled them, the region-default gets restored. But post join
(OTAA) that should not matter ...

Well, unless you have ADR enabled, in which case the gateway can re-enable
channels.

What should really be done is to disable channels before the join, so that
the join can be limited to the channels the gateway supports. If the
gateway later says it supports more, then that;s fine

If you want a single channel gateway, then this gateway should not enable
channel it does not support.

rossbug, 2018-09-12:
Hi 
I also found a similar problem.
My code is shown below. I am using the ABP not OTAA.
void setup( void )
{
   LoRaWAN.begin(EU868);
  for (unsigned int i = 3; i <16; i++)
 {  
    LoRaWAN.disableChannel(i);
    LoRaWAN.removeChannel(i);
 }
  LoRaWAN.setADR(false);
  LoRaWAN.setDataRate(0);
  LoRaWAN.joinABP(appEui, appKey, devEui);
  while(!LoRaWAN.joined()) {   };
}
void loop( void )
{
    if (!LoRaWAN.busy() && LoRaWAN.joined())
    {
        LoRaWAN.beginPacket(25);
        //-------------------
        use LoRaWAN.write() to write 9 data bytes
        //-------------------       
        LoRaWAN.endPacket();
    }
    STM32L0.stop(120000);
}

And i found a very strange thing on the TTN gateway as below picture.
[image]

At the same time, not only one channel is used to sand but also another channel with 867.7 or 867.5 MHz.
Does this problem is also the problem of LoRaWAN.disableChannel();

Thank you very much
Ross

GrumpyOldPizza, 2018-09-28:
Single Channel Gateways are not supported via hackery on the node side. To be region compliant your gateway needs to send it's proper channel mask post joinOTAA(). There is no way to override the gateways channel mask, as every downlink could change that. The modified LoRaMac-node code in use allows ADR without DR/TX_POWER change to change the channel mask if ADR is disabled. Again, the code for that is implemented. If you use joinABP(), then the node needs to enable/disable the proper channels.

closed 2018-09-28 by GrumpyOldPizza

nalberto, 2018-11-18:
How do you show which channel / frequency is being used to try to join the gateway in the serial console?
I would want to debug but not sure how to display that?

GrumpyOldPizza, 2018-11-19:
There is no way provided to track which channels are used in which sequence
for JOIN.

nalberto, 2018-11-19:
So do you practicaly check which frequencies are used when trying to join?

The join process takes sometime a very long time and I suspect it uses
channels out of range. Sometimes it just never joins with the same code
after resetting.

I have also tried disabling all channels and it still joins on the same
frequency 916.8 Australia TTN.

I feel totally blind. I have seen some code using lmic/Arduino that was
sending debugging info to serial with join frequencies.

GrumpyOldPizza, 2018-11-19:
I *think* there are 3 different region specifications applicable for
Australia. Do you know which one your gateway is using (yes, I know that
sounds like a stupid question, but others have run into that where AS923
was used instead of AU915).

Also if it's AU915, did you select the proper subband ?

GrumpyOldPizza, 2018-11-19:
> I use AU915 subband 2 which I understand is the right frequency for TTN
> Australia.
> I suspect it ignores the subband when joining and tries the whole set of
> channels randomly.

Actually it does not ignore the subband during join. If you don't specifiy
a subband it tries all 64+8 channels in a special sequence to touch all
subbands as quickly as possible.

But that goes back to the question, if you have a close by gateway, is it
really on AU915 ?

nalberto, 2018-11-19:
Yes gateway is on my roof and not that busy. It does join but very
inconsistent behavior. Could take 1 min to join on try for 40 min without
success on same code and close to gateway.

I cannot use this for low power device as the battery may drain on reset.

> Actually it does not ignore the subband during join. If you don't specifiy
> a subband it tries all 64+8 channels in a special sequence to touch all
> subbands as quickly as possible.
>
> But that goes back to the question, if you have a close by gateway, is it
> really on AU915 ?
>
>
> > I use AU915 subband 2 which I understand is the right frequency for TTN
> > Australia.
> > I suspect it ignores the subband when joining and tries the whole set of
> > channels randomly.
> >
> >
> > > I *think* there are 3 different region specifications applicable for
> > > Australia. Do you know which one your gateway is using (yes, I know
> that
> > > sounds like a stupid question, but others have run into that where
> AS923
> > > was used instead of AU915).
> > >
> > > Also if it's AU915, did you select the proper subband ?
> > >
> > >
> > > > So do you practicaly check which frequencies are used when trying to
> > > join?
> > > >
> > > > The join process takes sometime a very long time and I suspect it
> uses
> > > > channels out of range. Sometimes it just never joins with the same
> code
> > > > after resetting.
> > > >
> > > > I have also tried disabling all channels and it still joins on the
> same
> > > > frequency 916.8 Australia TTN.
> > > >
> > > > I feel totally blind. I have seen some code using lmic/Arduino that
> was
> > > > sending debugging info to serial with join frequencies.
> > > >
> > > >
> > > >
> > > >
> > > > > There is no way provided to track which channels are used in which
> > > > sequence
> > > > > for JOIN.
> > > > >
> > > > >
> > > > >
> >
> > > > >
> > > > > > How do you show which channel / frequency is being used to try to
> > > join
> > > > > the
> > > > > > gateway in the serial console?
> > > > > > I would want to debug but not sure how to display that?
> > > > > >
> > > > > > <
> > > > >
> > > >
> > >
> >
> https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439698067
> > > > > >,
> > > > > > <
> > > > >
> > > >
> > >
> >
> > > > > >
> > > > > >
> > > > >
> > > > > <
> > > >
> > >
> >
> https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439869580
> > > > >,
> > > > > <
> > > >
> > >
> >
> > > > >
> > > > >
> > > >
> > > > <
> > >
> >
> https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439886179
> > > >,
> > > > <
> > >
> >
> > > >
> > > >
> > >
> > > <
> >
> https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439888185
> > >,
> > > <
> >
> > >
> > >
> >
> > <
> https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439892696
> >,
> > <

Actually it does not ignore the subband during join. If you don't specifiy
a subband it tries all 64+8 channels in a special sequence to touch all
subbands as quickly as possible.

But that goes back to the question, if you have a close by gateway, is it
really on AU915 ?

> I use AU915 subband 2 which I understand is the right frequency for TTN
> Australia.
> I suspect it ignores the subband when joining and tries the whole set of
> channels randomly.
>
>
> > I *think* there are 3 different region specifications applicable for
> > Australia. Do you know which one your gateway is using (yes, I know that
> > sounds like a stupid question, but others have run into that where AS923
> > was used instead of AU915).
> >
> > Also if it's AU915, did you select the proper subband ?
> >
> >
> > > So do you practicaly check which frequencies are used when trying to
> > join?
> > >
> > > The join process takes sometime a very long time and I suspect it uses
> > > channels out of range. Sometimes it just never joins with the same
code
> > > after resetting.
> > >
> > > I have also tried disabling all channels and it still joins on the
same
> > > frequency 916.8 Australia TTN.
> > >
> > > I feel totally blind. I have seen some code using lmic/Arduino that
was
> > > sending debugging info to serial with join frequencies.
> > >
> > >
> > >
> > >
> > > > There is no way provided to track which channels are used in which
> > > sequence
> > > > for JOIN.
> > > >
> > > >
> > > >
> > > >
> > > > > How do you show which channel / frequency is being used to try to
> > join
> > > > the
> > > > > gateway in the serial console?
> > > > > I would want to debug but not sure how to display that?
> > > > >
> > > > > <
https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439698067
> > > > >,
> > > > > <
https://github.com/
> > > > >
> > > > >
> > > >
> > > > <
https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439869580
> > > >,
> > > > <
> > > >
> > > >
> > >
> > > <
https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439886179
> > >,
> > > <
> > >
> > >
> >
> > <
https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439888185

> >,
> > <

> >
> >
>
> <
https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/5#issuecomment-439892696