GrumpyOldPizza/ArduinoCore-stm32l0 #8: LoRaWAN payload limited to 11 bytes for any DataRate.

SELT, 2018-05-01:
It appears that the LoRa payload size is limited to 11 bytes for all data rates.  Tested specifically for data rates 2 and 3.

GrumpyOldPizza, 2018-05-01:
Do you have ADR enabled ? If so, then the stack will start off with DR_0
(i.e. 11 bytes for US915), or whatever you had set via
LoRaWAN.setDataRate() and then wait for the gateway to switch different
data rates that would allow more payload.

If you disable ADR, then you can set explicitly the data rate and hence get
larger payloads.

- Thomas

GrumpyOldPizza, 2018-05-01:
> Thomas:
>
> I have ADR disabled and am setting the DataRates explicitly. I can see
> from the server that the SF and BW are as expected, but The Tx fails for
> payload sizes > 11.
>
> Jon

Ack. With OTAA join, the stack forces the data rate to DR_0 for join, and
overrides the user selection. It was supposed to revert back to the default
(i.e. your override), but doesn't.

SELT, 2018-05-01:
Thomas:

Looks like the DataRate reverts back OK, just not the ma payload size.

Jon

GrumpyOldPizza, 2018-05-01:
> GrumpyOldPizza/ArduinoCore-stm32l0/issues/8#issuecomment-385722890>, or

Not quite. There is a pre-check as to whether the size of the packet will
fit. And that one fails because the data rate does not come back in time.

GrumpyOldPizza, 2018-05-04:
Fixed. Issue was at the end of that day that LoRaMacQueryTxPossible() had no way to consider a user supplied datarate.

closed 2018-05-04 by GrumpyOldPizza