GrumpyOldPizza/ArduinoCore-stm32l0 #9: STM32 BOOTLOADER not showing in Zadig

HWittlich, 2018-05-07:
First of all, thank you for your time!
While following the step-by-step guide on a win7 machine, I encountered the following problem:
When showing all devices in Zadig, I cant see a STM32 BOOTLOADER. Instead, the B-L072Z-LRWAN1 board shows up as ST-Link mass storage, ST-Link Debug and ST-Link VCP Ctrl.
[image]

I'm unsure how to proceed from here, your help would be greatly appreciated.

GrumpyOldPizza, 2018-05-07:
The STM32 BOOTLOADER / zadig routine is only needed for boards that
directly use USB. Nucleo boards use the ST-Link Debug Interface. Just
install the drivers off ST's webpage.

HWittlich, 2018-05-08:
Ah, thank you so much for your help. 
Just in case someone else happens to look for the drivers, I found them here:

[ST-LINK, ST-LINK/V2, ST-LINK/V2-1 USB driver signed for Windows7, Windows8, Windows10](http://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-utilities/stsw-link009.html)

closed 2018-05-08 by HWittlich