GrumpyOldPizza/ArduinoCore-stm32l0 #10: OpenOCD init failed

HWittlich, 2018-05-08:
Dear GrumpyOldPizza,

I tried to upload a Sketch to my B-L072Z-LRWAN1 board, (chose right board and COM port in Arduino).
However, I get the following error message:

```
An error occurred while uploading the sketch
Error: init mode failed (unable to connect to the target)
in procedure 'program' 
in procedure 'init' called at file "embedded:startup.tcl", line 495
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked
```
Any idea what I could do? Thank you for your help.

GrumpyOldPizza, 2018-05-08:
Sometimes the STM32L0 chip is in a low power mode, where openocd cannot
connect. Hold down the RESET button while uploading the sketch.

HWittlich, 2018-05-08:
I am amazed by your in-depth knowledge about all of this... Really incredible.

At first, while holding reset down I got `Error: timed out while waiting for target halted` and Windows 7 told me that openOCD.exe stopped working. Then I tried uploading again, releasing the button shortly after starting the upload. Voila! the code was uploaded and the boards LED turned green. 

Now I just have to figure out why LoRaWAN.joined() returns 0 and doesn't connect to TTN. Perhaps I will have to give it another try outdoor.

closed 2018-05-08 by HWittlich

GrumpyOldPizza, 2018-05-08:
TTN. What region ? Are you registered properly ?

HWittlich, 2018-05-08:
EU Region, there should be at least 3 Gateways in reach. I'm using the example sketch from LoRaWAN.h and have entered my device/app EUIs and appkey from the TTN console. Is there more I need to do to register properly?