GrumpyOldPizza/ArduinoCore-stm32l0 #12: B-L072Z-LRWAN1 Board not connecting when running on batteries

HWittlich, 2018-05-22:
Dear GrumpyOldPizza,

I'm using your example code, and as long as the board is connected power via USB, the LEDs (LD5, LD2) are lit up and the board correctly connects to TTN. However, if I put three AAA batteries in the battery compartment, only LD2 lights up and the board does not connect to TTN. I am clueless as to why that is. Any ideas? I greatly appreciate your help!

nicbkw, 2018-06-06:
Note that ST-Link portion of the DISCO board affects RESET of the Murata module and is not powered by battery, so it is necessary to remove SB37, explained at the bottom of page 14 of http://www.st.com/content/ccc/resource/technical/document/user_manual/group0/ac/62/15/c7/60/ac/4e/9c/DM00329995/files/DM00329995.pdf/jcr:content/translations/en.DM00329995.pdf

nicbkw, 2018-06-06:
You are right, but it is possible to flash the module after removing SB37 via St-Link by pressing the Black reset switch until you see the start of the  flash process. It is then necessary to press the same reset button  to launch the code. It is awkward and I think I might add a switch in place of SB37...

GrumpyOldPizza, 2018-06-06:
Nick, you got the right idea about the problem. On the other hand, what if you add a 100k pullup on RESET to VCC via the external header. Shouldn't this work as well ?

What I don't quite get is why you need to remove SB37. There is no pulldown on T_NRST. If one uses battery power then the ST-LINK part is not powered, hence T_NRST on PB3 should be analog input without pullup or pulldown ...

GrumpyOldPizza, 2018-06-06:
CN2 has MCU_nRST, and 3V3 is right next to it. So a simple jumper should work, after you flashed the code ?

GrumpyOldPizza, 2018-06-08:
Putting a jumper between NRST and 3V3 on CN2 makes B-L072Z-LRWAN1 work with battery. However I do see a bunch of LEDs still active, so that may not be the best setup to test battery life.

closed 2018-06-08 by GrumpyOldPizza