GrumpyOldPizza/ArduinoCore-stm32l0 #13: Interrupts

rmeldo, 2018-06-02:
Hi, 
Do the interrupts work?
I am trying to find out which pins are enabled for using them.

Thank
Riccardo

GrumpyOldPizza, 2018-06-02:
Interrupts work.

All Dxx pins allow interrupts. I need to come up with some better
documentation there though.

- Thomas

rmeldo, 2018-06-03:
Thanks, 
I found out that the High and low options are not supported, whereas rising, falling and change work, and they are what's needed most of the time.
Riccardo

GrumpyOldPizza, 2018-06-06:
Issue is that LOW and HIGH are not supported for STM32L0.

The semantics are that a interrupt get triggered as long as a pin has the selected level. Hence at enable time the code had to check (and pend the ISR) the GPIO. At the tail of the ISR handler one had to also check the GPIO and if the level is still matching, re-pend the ISR.

Try to get this added in a bit. IMHO this is not really that useful I'd assume, but it should be there for compatibility.

rmeldo, 2018-06-06:
I see your point and agree.
Once you are in the ISR you can keep checking the level until you're sure
it is low and act accordingly, or exit without acting.

I also agree that for compatibility it is desirable to have the High and
Low options.

Riccardo

ucdos123456, 2018-10-17:
@GrumpyOldPizza 
Hi. i redefine D1 to PA13. Do this pin supports interrupt?

GrumpyOldPizza, 2018-10-17:
PA13 supports interrupts. But PA13 is also a SWD pin. So you lose the ability to debug the system ...

paengpd, 2019-06-17:
Hi,
' All Dxx pins allow interrupts. '
Other pins are not work? Only Dxx pins are available?
ex) A0~A4, SCL, SDA...

GrumpyOldPizza, 2019-06-17:
A0 to A4 should work as well, plus/minus aliasing (PA2 would alias with PB2, which means only one of them can be an interrupt). Do not use interrupts on SCL/SDA, ever. They are I2C pins, sometimes with internal peripherals mapped.

closed 2019-08-30 by rmeldo

hallard, 2019-12-11:
Yes I confirm, even with no I2C device and not enabled with `wire`, on grasshopper IRQ on SDA works but not on SCL

GrumpyOldPizza, 2019-12-11:
My bad. Need to fix the variant.cpp files. PA8 aliases PB8, so PB8 cannot have a interrupt. Same goes with PA9 and PB9. So neither SDA nor SCL should have interrupts on them.

hallard, 2019-12-11:
It's no problem, I wanted to use my 2 I2C exposed pin as IRQ for a demo without changing anything on my board. So I used another pin as other Interrupt but I have no more left, so I'm happy SDA works ;-)

sanketa-54, 2022-04-01:
Hi
Can we use attachInteruupts on pin PA13 and PA14

GrumpyOldPizza, 2022-04-02:
Yes, but not recommended, as PA13/PA14 are SWD.

sanketa-54, 2022-04-04:
Ok...Thanks for reply.

sanketa-54, 2022-04-04:
Also PB13 and PB14 are supported for attachInterrupt() ??
if yes , is there anything to be done in variant.cpp file