GrumpyOldPizza/ArduinoCore-stm32l0 #15: OTAA join data rate sequence

GrumpyOldPizza, 2018-06-21:
LoRaMac-node injects the datarate for a single JOIN into MlmeReqJoin_t. Hence the LoRaWAN class will only use the currently configured datarate for repeated joins.

Implement a scheme where after the initial sequence throu the join channels, that the next lower data rate is being used. This way a EU868 use can start with SF7 and escalate to SF12.

Perhaps the join should also ALWAYS use the highest txpower, no matter what the user selected. Only post join the user datarate / txpower are set for the stack.

GrumpyOldPizza, 2018-07-17:
> Hi,
>
> I'm experimenting LoRaWAN with your library and Murata ABZ. I'm in France
> so I'm using EU868 and effectively I observed that it use SF12, even if I'm
> 10 meters from my gateway. And logically it leads to an airtime ~800ms when
> my mDot in SF7 takes ~ 30 ms.
>
> So what you're telling in your post is that I should play with
> setDataRate(unsigned int datarate) and
> setTxPower(float power) to get the best energetic performance ?

The stack defaults to SF12 and ADR. So your gateway can lower TxPower and
the increase the Datarate.

The alternative is that you are disable ADR and set you own
TxPower/Datarate.

battosai30, 2018-07-17:
I was about to told you that ADR seems not to work, but after 64 packets the datarate shift from 0 to 5 (so SF7BW125) and TxPower from 16 dbm to 12 dbm. So apparently everythings is good and I checked LoRaWAN specification and it's exactly that : 

_An ADR request is sent when the device sets the ADRAckReq bit. By default this happens after sending 64 uplinks without receiving a downlink, but as that depends on the device implementation we can’t give you an exact number here._

Thanks again for these precisions

battosai30, 2018-07-27:
I tried to make my own dynamic adaptation : tuning SF works well, but Txpower seems not to be applied.

#include "LoRaWAN.h"
#include "STM32L0.h"

const char *devAddr = "xxx";
const char *nwkSKey = "xxx";
const char *appSKey = "xxx";

byte SF=2;
byte TXP=16;

void setup( void )
{

  Serial.begin(9600);
  while (!Serial);

  LoRaWAN.begin(EU868);

  LoRaWAN.disableChannel(0);
  LoRaWAN.disableChannel(1);
  LoRaWAN.disableChannel(2);
  LoRaWAN.disableChannel(3);

  LoRaWAN.addChannel(0, 868100000, 0, 7);
  LoRaWAN.enableChannel(0);
  LoRaWAN.setRX2Channel(869525000, 3);

  LoRaWAN.setDataRate(SF);
  LoRaWAN.setTxPower(TXP);
  LoRaWAN.joinABP(devAddr, nwkSKey, appSKey);

  Serial.println("JOIN( )");
}

void loop( void )
{
  if (LoRaWAN.joined() && !LoRaWAN.busy())
  {

    if (LoRaWAN.confirmed())
    {
      Serial.println("ACK");
    }
    else
    {
      Serial.println("NACK");
    }

    Serial.print("CHECK( ");
    Serial.print("RSSI: ");
    Serial.print(LoRaWAN.lastRSSI());
    Serial.print(", SNR: ");
    Serial.print(LoRaWAN.lastSNR());
    Serial.print(", Margin: ");
    Serial.print(LoRaWAN.linkMargin());
    Serial.print(", Gateways: ");
    Serial.print(LoRaWAN.linkGateways());
    Serial.println(" )");

if(LoRaWAN.lastRSSI()<(-10) && LoRaWAN.lastRSSI()>(-80)) {

  if(SF<5){
   SF++;
  LoRaWAN.setDataRate(SF);
   Serial.print("New DR = ");
   Serial.println(SF);
  } else {
    if(TXP>2) TXP-=2;
    LoRaWAN.setTxPower(TXP);
     Serial.print("New TXP = ");
   Serial.println(TXP);
   LoRaWAN.joinABP(devAddr, nwkSKey, appSKey);
  }

}
    Serial.print("TRANSMIT( ");
    Serial.print("TimeOnAir: ");
    Serial.print(LoRaWAN.getTimeOnAir());
    Serial.print(", NextTxTime: ");
    Serial.print(LoRaWAN.getNextTxTime());
    Serial.print(", MaxPayloadSize: ");
    Serial.print(LoRaWAN.getMaxPayloadSize());
    Serial.print(", DR: ");
    Serial.print(LoRaWAN.getDataRate());
    Serial.print(", TxPower: ");
    Serial.print(LoRaWAN.getTxPower(), 1);
    Serial.print("dbm, UpLinkCounter: ");
    Serial.print(LoRaWAN.getUpLinkCounter());
    Serial.print(", DownLinkCounter: ");
    Serial.print(LoRaWAN.getDownLinkCounter());
    Serial.println(" )");

    LoRaWAN.beginPacket();
    LoRaWAN.write(0xef);
    LoRaWAN.write(0xbe);
    LoRaWAN.write(0xad);
    LoRaWAN.write(0xde);
    LoRaWAN.endPacket(true);

  }

  delay(10000);
  Serial.println("tick");
}`

TXP variable is well updated, but  Serial.print(LoRaWAN.getDataRate()) always output 16 dbm.