GrumpyOldPizza/ArduinoCore-stm32l0 #16: STM32L0.sleep() and STM32L0.stop() not working on B-L072Z-LRWAN1

jaimelaborda, 2018-06-23:
I'm using a B-L072Z-LRWAN1 but methods sleep() and stop() of the STM32L0 library seems to be not working. It compiles without error, but the programs just keeps running no matter what value you put into the methods.

My code:

```
#include <STM32L0.h>
#include <Adafruit_SHT31.h>
#include <Adafruit_TSL2591.h>
#include "LoRaWAN.h"

const char *appEui = "XXXXXXXXXXXXXXXX";
const char *appKey = "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX";
const char *devEui = "XXXXXXXXXXXXXXXX";

#define LPP_DATATYPE_TEMPERATURE 0x67
#define LPP_DATATYPE_HUMIDITY 0x68
#define LPP_DATATYPE_ILLUMINANCE 0x65

Adafruit_SHT31 sht31;
Adafruit_TSL2591 tsl = Adafruit_TSL2591(2591);

void setup( void )
{

    Serial.begin(115200);
    sht31.begin(0x44);
    
    tsl.begin();
    tsl.setGain(TSL2591_GAIN_MED);
    tsl.setTiming(TSL2591_INTEGRATIONTIME_100MS);
    
  
    LoRaWAN.begin(EU868);
    LoRaWAN.addChannel(3, 867100000, 0, 5);
    LoRaWAN.addChannel(4, 867300000, 0, 5);
    LoRaWAN.addChannel(5, 867500000, 0, 5);
    LoRaWAN.addChannel(6, 867700000, 0, 5);
    LoRaWAN.addChannel(7, 867900000, 0, 5);
    LoRaWAN.setDataRate(5);
    LoRaWAN.joinOTAA(appEui, appKey, devEui);
}

void loop( void )
{
    
    uint16_t temp = (uint16_t) (sht31.readTemperature() * 10);
    uint16_t rh = (uint16_t) (sht31.readHumidity() * 2);
    Serial.println("Temp: "+String(temp/10));
    Serial.println("RH:   "+String(rh/2));

    uint32_t lum = tsl.getFullLuminosity();
    uint16_t ir, full;
    ir = lum >> 16;
    full = lum & 0xFFFF;
    uint16_t lux = tsl.calculateLux(full, ir);
    Serial.println("Lux:   "+String(lux));

    if (!LoRaWAN.busy() && LoRaWAN.joined())
    {
        int lpp_channel = 0;
        LoRaWAN.beginPacket();
        LoRaWAN.write(lpp_channel++);
        LoRaWAN.write(LPP_DATATYPE_TEMPERATURE);
        LoRaWAN.write((temp >> 8) & 0xFF);
        LoRaWAN.write(temp & 0xFF);
        LoRaWAN.write(lpp_channel++);
        LoRaWAN.write(LPP_DATATYPE_HUMIDITY);
        LoRaWAN.write(rh & 0xFF);
        LoRaWAN.write(lpp_channel++);
        LoRaWAN.write(LPP_DATATYPE_ILLUMINANCE);
        LoRaWAN.write((lux >> 8) & 0xFF);
        LoRaWAN.write(lux & 0xFF);
        LoRaWAN.endPacket();
    }

    STM32L0.stop(5000);
}
```

Test it with a simple sketch and it works sleep() either stop() methods:

```
#include <STM32L0.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

}
void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("Wake!");
  STM32L0.sleep(30000);
}

```
Cannot do it works with the library LoRaWAN. Maybe due to OTAA join? I don't know if it has a lot of sense. I tried also to comment the sensors libraries and appears to be the LoRaWAN library.

Thank you!

GrumpyOldPizza, 2018-06-23:
Mind trying that simple code below quickly (verbatim) ?

```
void setup( void )
{
    Serial.begin(9600);
    
    while (!Serial) { }

    LoRaWAN.begin(EU868);
    LoRaWAN.setDutyCycle(false);
    LoRaWAN.joinOTAA(appEui, appKey, devEui);

    Serial.println("JOIN( )");
}

void loop( void )
{
    Serial.println("top-of-loop");
    
    if (LoRaWAN.joined() && !LoRaWAN.busy())
    {
        Serial.println("TRANSMIT( )");

        LoRaWAN.beginPacket();
        LoRaWAN.write(0xef);
        LoRaWAN.write(0xbe);
        LoRaWAN.write(0xad);
        LoRaWAN.write(0xde);
        LoRaWAN.endPacket();
    }

    STM32L0.stop(10000);
}
```
For me that work here. If it does not for you ... is there a chance that you are running some older code ? How did you install the core, via json file or a down load of the github ?

jaimelaborda, 2018-06-25:
Hi GrumpyOldPizza,

It works!

I was using a older version from May. Just update and it works. 

Thanks for your herlp!

closed 2018-06-25 by jaimelaborda