GrumpyOldPizza/ArduinoCore-stm32l0 #17: Linux path issue

SloMusti, 2018-07-07:
I have installed this repository as instructed on Ubuntu and ran into the following error:

```/home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.5/tools/linux//dfu-suffix: no such file or directory
Error compiling for board Cricket-L082CZ.
```
The files are confirmed in the right place, however there is a redundant / in the path, which I can not figure out where from.

GrumpyOldPizza, 2018-07-07:
Can you append a screen shot of the error log ?

GrumpyOldPizza, 2018-07-08:
> Full Arduino debug output on setting with all messages:
>
> /home/musti/Projects/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/musti/Projects/arduino-1.8.5/hardware -hardware /home/musti/.arduino15/packages -hardware /home/musti/Arduino/hardware -tools /home/musti/Projects/arduino-1.8.5/tools-builder -tools /home/musti/Projects/arduino-1.8.5/hardware/tools/avr -tools /home/musti/.arduino15/packages -built-in-libraries /home/musti/Projects/arduino-1.8.5/libraries -libraries /home/musti/Arduino/libraries -fqbn=TleraCorp:stm32l0:Cricket-L082CZ:usb=cdc,dosfs=none,speed=32,opt=os -vid-pid=0X2341_0X8053 -ide-version=10805 -build-path /tmp/arduino_build_533590 -warnings=all -build-cache /tmp/arduino_cache_307305 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/musti/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -verbose /home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.6/libraries/LoRaWAN/examples/LoRaWAN_Join_ABP/LoRaWAN_Join_ABP.ino
> /home/musti/Projects/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/musti/Projects/arduino-1.8.5/hardware -hardware /home/musti/.arduino15/packages -hardware /home/musti/Arduino/hardware -tools /home/musti/Projects/arduino-1.8.5/tools-builder -tools /home/musti/Projects/arduino-1.8.5/hardware/tools/avr -tools /home/musti/.arduino15/packages -built-in-libraries /home/musti/Projects/arduino-1.8.5/libraries -libraries /home/musti/Arduino/libraries -fqbn=TleraCorp:stm32l0:Cricket-L082CZ:usb=cdc,dosfs=none,speed=32,opt=os -vid-pid=0X2341_0X8053 -ide-version=10805 -build-path /tmp/arduino_build_533590 -warnings=all -build-cache /tmp/ardu [... cut ...]
> Using library LoRaWAN at version 1.1.0 in folder: /home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.6/libraries/LoRaWAN
> fork/exec /home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.6/tools/linux//dfu-suffix: no such file or directory
> Error compiling for board Cricket-L082CZ.

Got a fix for this, I'd think. The "//" might be the issue. Not sure why it
would be, because this works on my system quite well.

Can you check you have
a /home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.6/tools/linux/dfu-suffix
and that it's executable (and readable) ?

SloMusti, 2018-07-08:
I have fug a bit deeper and found these are 32-bit executables, checked with:

```
file dfu-prefix 
dfu-prefix: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=e4dc89b4522cebf6e62294585516e4c7bb6e8fdb, not stripped
```
following this instructions: https://askubuntu.com/a/454452 I have installed 32-bit support and now it works well. This issue can be closed, but would be great to add the note to instructions.

GrumpyOldPizza, 2018-07-14:
Switched to 64 bit binaries. Fixed.

closed 2018-07-14 by GrumpyOldPizza