GrumpyOldPizza/ArduinoCore-stm32l0 #20: Program via serial port

Protoneer, 2018-07-10:
I have a Grasshopper board and to program it via Arduino I need to hold the boot button while restarting the board. 

Can this board be programmed without having to put it in bootload mode manually?

I am also not able to get any of the serial examples to work. Do use Serial or SerialUSB? Tried both with no luck.

I am currently programming it from Win10. 

I am able to get the blink sketch to work. Any help will be much appreciated.

GrumpyOldPizza, 2018-07-10:
Does any of the examples that print out bits and pieces show any output on
the serial monitor ?

Protoneer, 2018-07-10:
No, the board does not expose a Serial Port so I can't open the serial monitor.

GrumpyOldPizza, 2018-07-10:
Grasshopper has a USB port, which serves as "Serial". So the serial monitor
in the Arduino IDE should allow you to see data that a sketch prints out.

- Thomas

Protoneer, 2018-07-10:
Yes... that was my understand as well.

From the picture below you can see there are no port selectable and that is why the Serial Monitor is not opening.

[image]

I do get an error message when flashing the board.(Manually have to put it into bootload mode) Is that normal?
[image]

GrumpyOldPizza, 2018-07-10:
I had never seen this error before.

If you look in the device manager then, is there another USB port after you
flashed this ?

Perhaps using Zadig, does it show up ?

- Thomas

Protoneer, 2018-07-10:
Unfortunately no serial port...
[image]

When I put it into bootloader mode I do get a STM32 BOOTLOADER device...
[image]

I tried Zadig....
[image]

Mmmmmm... wonder what it could be.

GrumpyOldPizza, 2018-07-10:
Ok, in the images you selected a "Grasshopper" with 32MHz clock. You do
have a real Grasshopper board, right ?

I am puzzled as STM32 BOOTLOADER works, so it's not a hardware issue (bad
cable, busted board, bad USB port ...). You also indicated that the Blink
sketch makes the board blink, which tells me you most likely did not select
the wrong board. It's also not stuck in the bootloader ... Your screen-shot
shows a 32MHz clock, which allows USB to work (16MHz and lower disables
USB).

I have no idea what could be going on.

Could you try something like this here to list the USB devices:
https://www.nirsoft.net/utils/usb_devices_view.html

Protoneer, 2018-07-10:
Managed to get DevManView... It still picks up when in bootloader mode but not in normal mode.

Yes its a real Grasshopper board I bought of Tindie. 

Does these boards come pre-programmed with special Bootloader firmware that I might have erased? Like the Arduino Zero...

Yes....I have assembled the Grasshopper from OSHpark PCB's...

GrumpyOldPizza, 2018-07-10:
The bootloader is in ROM, factory programmed. You cannot nuke that one.

Perhaps you got the wrong PCB, or made a soldering mistake. There is a USB
connect pin via PA8.

Mine editing manually the variant.h file for Grasshopper and change the
following line:

#define STM32L0_CONFIG_PIN_VBUS           STM32L0_GPIO_PIN_NONE

Protoneer, 2018-07-10:
Right... That did the trick!!! 

Is there a reason for having the dedicated pin?

Is there a way that I can make a $$$ donation for your help?

closed 2018-07-10 by Protoneer

GrumpyOldPizza, 2018-07-11:
The reason to have a pin is that the USB peripheral needs quite a bit of
power when turned on an waiting to connect. It's easier to monitor the
voltage.