GrumpyOldPizza/ArduinoCore-stm32l0 #21: PB6 digital write issue ?

battosai30, 2018-07-16:
Hi !
Finally received my boards with Murata ABZ module. I have a strange problem : the led I attached on pin PB6 doesn't light on. I checked with my voltmeter , nothings comes out. With pins around it, no problem. I checked in variant.cpp and nothing seems wrong.

Before going deeper, if you have any idea I take it ... Or if someone could check on his board and confirm it just mine which has the problem he's welcome :)

Regards

GrumpyOldPizza, 2018-07-16:
There should be nothing that prevents PB6 from working ....

- Thomas

battosai30, 2018-07-17:
Yep you are right. 3 boards soldered, exactly only this pin that was bugging and it was a soldering issue. So much unlucky ... But when I desoldered it, I observed that some pad doesn't seems to "like" tin (maybe fat traces). So I will clean it up before soldering now ...

Thanks

closed 2018-07-17 by battosai30