GrumpyOldPizza/ArduinoCore-stm32l0 #22: [LoRaWAN] First packet is empty

battosai30, 2018-07-17:
Hi,

Every time I run for example LORAWAN_TTN_OTAA.ino , the first packet is empty : 

[image]

I have not this issue with my mDot868.

GrumpyOldPizza, 2018-07-17:
That is correct.

We do send an empty packet to give the gateway a chance to update the
network parameters right away. It uses "port 0" for that. That is mainly
required for AU915/US915, where there is no CFLIST for OTAA.

battosai30, 2018-07-17:
Ok so the empty packet is "included" with the join sequence and it's not a : 
```
   LoRaWAN.beginPacket();
        LoRaWAN.write(0xef);
        LoRaWAN.write(0xbe);
        LoRaWAN.write(0xad);
        LoRaWAN.write(0xde);
        LoRaWAN.endPacket();
```
That's "ignored" ?

GrumpyOldPizza, 2018-07-17:
>  That's "ignored" ?

I has no payload and goes to port 0. So it's not a missing/bad application
data packet.

battosai30, 2018-07-17:
Perfect !
Thanks

closed 2018-07-17 by battosai30