GrumpyOldPizza/ArduinoCore-stm32l0 #23: [LoRaWAN] hangs up

battosai30, 2018-07-17:
Hi,

Some times with the LORAWAN_TTN_OTAA example, the system just hangs up randowly : 

[image]

In the screenshot I just some serial debug informations and a blink led but did not changed anything else. With the blink led I know that all the system is hanging because it freezes too. The bug happens all the time but it stops at differents steps (sometimes after "TRANSMIT (" sometimes avec "DR : 0" ...). 

Does it happen to anyone else ? Except bug in library, it could be a supply issue I guess but I doubt ... I will check that by passing the actual linear regulator and connecting a high performance supply.

My config : 
- Own board with murata ABZ + BME 280.
- Europe 868 MHz
- Real gateway with Rpi3 + ic880a connected to TTN

GrumpyOldPizza, 2018-07-17:
Please upgrade to 0.0.7. There was a nasty SVC/ISR register corruption
issue that I never could reproduce till last week.

battosai30, 2018-07-17:
You're a chief ! It still to  work well know ;)

Big thanks !

closed 2018-07-17 by battosai30