GrumpyOldPizza/ArduinoCore-stm32l0 #24: Support for Sigfox connectivity

kbesson, 2018-07-17:
Hi,

First, thank you for your amazing work ! :)

Are you planning to add Sigfox connectivity on Arduino for the Murata module ?

I can't find any native implementations of the Sigfox stack for Arduino. Only examples I've found are for the MKRFOX1200 that use a Sigfox modem with AT command set.

There are few examples projects in the Sigfox STCubeMX package but I don't think I will be able to port the stack for Arduino...

Thanks !

GrumpyOldPizza, 2018-07-18:
There is nothing tentative planned at the moment. I am not in an area where
I could actually test SigFox. So it's hard to write software then.

kbesson, 2018-07-23:
I understand, thanks for the quick reply @GrumpyOldPizza !

For information, you can check the Sigfox network coverage here : https://www.sigfox.com/en/coverage

And you can have free SigFox connectivity in the covered areas by requesting Murata for PAC files.

GrumpyOldPizza, 2018-07-27:
Kevin, can you contact me offline via grumpyoldpizza@gmail.com ?

kbesson, 2018-07-27:
Done ! ;)

GrumpyOldPizza, 2018-08-30:
SigFox will not be supported in the current hardware. The other issue is the SigFox certification requirement. So ST did that for their B-L072Z-LRWAN1 platform. It seems to be impractical for now to do that for any other supported board ...

closed 2018-08-30 by GrumpyOldPizza

nobodyguy, 2018-12-31:
Hello @GrumpyOldPizza - thank you for your amazing work and dedication!
What do you mean by "current hardware"? Sigfox is officially supported in the Murata chip. And the certification is not needed for development purposes.

GrumpyOldPizza, 2018-12-31:
SigFox with the ST code requires some special GPIO mapping. We do not use this mapping as LoRaWAN is the prime target,

nobodyguy, 2018-12-31:
Understood, thank you for the explanation.