GrumpyOldPizza/ArduinoCore-stm32l0 #25: delayMicroseconds() timing is not good

battosai30, 2018-07-18:
Hi,

This time I think I've found a real bug xD

Trying to connect a DS18B20 (so OneWire protocol) I failed. I took my logic analyzer and surprise, timings are not good ! 

I used a simple code like this : 
```
    digitalWrite(2, LOW);
    delayMicroseconds(10);
    digitalWrite(2, HIGH);
    delay(20);

    digitalWrite(2, LOW);
    delayMicroseconds(50);
    digitalWrite(2, HIGH);
    delay(20);
    
    digitalWrite(2, LOW);
    delayMicroseconds(100);
    digitalWrite(2, HIGH);
    delay(20);

    digitalWrite(2, LOW);
    delayMicroseconds(480);
    digitalWrite(2, HIGH);
    delay(20);

    digitalWrite(2, LOW);
    delayMicroseconds(800);
    digitalWrite(2, HIGH);
    delay(20);

    digitalWrite(2, LOW);
    delayMicroseconds(1000);
    digitalWrite(2, HIGH);
    delay(20);

    digitalWrite(2, LOW);
    delayMicroseconds(1200);
    digitalWrite(2, HIGH);
    delay(20);
```
And with an excel sheet I calculated a correction that I applied to all timings in OneWire library and magic I got my DS18B20. 

So for example if you want a delay of 200µs you have to use delayMicroseconds(200*1.32-5) to get a real 200µs pause.

GrumpyOldPizza, 2018-07-18:
What CPU clock are you using ?

battosai30, 2018-07-18:
32 MHz did not try with the others but as I need USB for debug I do not really have the choice :s

GrumpyOldPizza, 2018-07-18:
I'll take a peek. Will take a bit. The loop is a software loop, which is
bad ... Should be replaced by a Systick based one to compensate for
prefetch deltas ...

battosai30, 2018-07-19:
Is there a particular reason to don't use the same approach as standard Arduino like Uno with a simple loop with a while comparing the target time with micros() ? I see in your armv6m_systick.h that a armv6m_systick_micros() exists and as I understand it it's based on systick interruption so it should a good timing reference no ?

GrumpyOldPizza, 2018-07-19:
The main reason (besides that all other cores than Arduino AVR) use the
same technique) is that the code is reused in the system layer, where
SysTick (which drivers micos()) is not available yet, or precise enough.

GrumpyOldPizza, 2018-08-30:
Fixed as of 08/30

closed 2018-08-30 by GrumpyOldPizza