GrumpyOldPizza/ArduinoCore-stm32l0 #26: [LoRaWAN] Single channel gateway issue

battosai30, 2018-07-24:
Hi,

I tried two single channel gateway and having troubles with both, but no problem on a "real" gateway (Rpi+ic-880). 

I tried OTAA and ABP.

OTAA : the node doesn't not seems to receive the JOIN_ACCEPT message. I tried a lot of config, the join resquest is well received (I have to fix DR to 5) the gateway seems to send well the join accept but the node  seems to receive nothing. 

ABP : it works but only ~1msg/5 succesfully arrived on server :s 

As the issue appears on two different gateways (one is esp32 + sx1276 board, the other is a Dragino LG01) I think it's not coming from the gateway firmwares or it's a protocol issue I missed something configuring node and/or gateway.

So if you have any suggestion :)

battosai30, 2018-07-25:
Ok I think I defeated the beast xD

So I used my mDot-868 to check if the problem comes from the gateway or the node. So, I decided to work on ABP mode first to eliminate downlink as a potential error source.

So first I observed the same troubles. But quickly I noticed something as mDot output more debug information, and espacially frequency used : frequencies are rolling. I don't know in details LoRaWAN protocol so I think its normal to reduce dutycycle. But with a mono-canal gateway it becomes a problem ... So I configured my mdot to only use 868.1 MHz and bingo, I received almost all packet (sometimes one is lost but I think it's the gateway that is used by another node). So on my Murata ABZ board I used this code : 

```
#include "LoRaWAN.h"

const char *devAddr = "******";
const char *nwkSKey = "******";
const char *appSKey = "******";

void setup( void )
{

  Serial.begin(9600);
  while(!Serial);
  
    LoRaWAN.begin(EU868);
    LoRaWAN.disableChannel(0);
    LoRaWAN.disableChannel(1);
    LoRaWAN.disableChannel(2);
    LoRaWAN.disableChannel(3);
    LoRaWAN.addChannel(0, 868100000, 0, 7);
    LoRaWAN.enableChannel(0);

    LoRaWAN.setAntennaGain(2.0);
    LoRaWAN.setDataRate(5);
    LoRaWAN.setTxPower(14);
    LoRaWAN.joinABP(devAddr, nwkSKey, appSKey);

    Serial.println("JOIN( )");
}

void loop( void )
{
    if (LoRaWAN.joined() && !LoRaWAN.busy())
    {
        Serial.print("TRANSMIT( ");
        Serial.print("TimeOnAir: ");
        Serial.print(LoRaWAN.getTimeOnAir());
        Serial.print(", NextTxTime: ");
        Serial.print(LoRaWAN.getNextTxTime());
        Serial.print(", MaxPayloadSize: ");
        Serial.print(LoRaWAN.getMaxPayloadSize());
        Serial.print(", DR: ");
        Serial.print(LoRaWAN.getDataRate());
        Serial.print(", TxPower: ");
        Serial.print(LoRaWAN.getTxPower(), 1);
        Serial.print("dbm, UpLinkCounter: ");
        Serial.print(LoRaWAN.getUpLinkCounter());
        Serial.print(", DownLinkCounter: ");
        Serial.print(LoRaWAN.getDownLinkCounter());
        Serial.println(" )");

        LoRaWAN.beginPacket();
        LoRaWAN.write(0xef);
        LoRaWAN.write(0xbe);
        LoRaWAN.write(0xad);
        LoRaWAN.write(0xde);
        LoRaWAN.endPacket();

    }

    delay(10000);
}
```
I don't know if I did the right way (I think some code is not need, advices are welcome :) )but at least it works.

Is there any way to get the frequency used for current transmit ?

Regards

GrumpyOldPizza, 2018-07-25:
Sorry to be so slow. Seems you figured it out.

There is no way to get the next frequency (or the current one). Normally
you have 3 base frequencies in LoRaWAN and the node picks one randomly that
is available (might be restricted due to duty cycling).

The APIs like LoRaWAN.getDataRate() are returning info about the NEXT
transmission, where things might change due to ADR.

In general, I'd recommend actually reading the spec before guessing ...
https://www.lora-alliance.org/resource-hub

battosai30, 2018-07-25:
> Sorry to be so slow. Seems you figured it out.
One day is not at all what I would call slow xD 

> In general, I'd recommend actually reading the spec before guessing ...
https://www.lora-alliance.org/resource-hub

I will take the time to read it well now that I have tested a lot of things and have a clearer view on LoRaWAN. 

Just one last point : I did not find where I could activate ack request ?

battosai30, 2018-07-25:
So simple ... And if endPacket(true) is used, I just have to do something like this ?

```
 while (LoRaWAN.busy())
    {
    }

    if (LoRaWAN.confirmed())
    {
      debug.println("ACK");
    }
    else
    {
      debug.println("NACK");
    }
```

GrumpyOldPizza, 2018-07-25:
You don't have to check ... But then you don't know whether it succeeded
or not ...

battosai30, 2018-07-26:
Ok so what could I do in ABP mode to check at least that a gateway is in the range ? At least I would check the link during start up

GrumpyOldPizza, 2018-07-26:
Link check does not need to be answered right away. A confirmed
transmission does.

closed 2018-09-28 by GrumpyOldPizza