GrumpyOldPizza/ArduinoCore-stm32l0 #28: booting takes 4-5 seconds

tve, 2018-08-03:
If I write a sketch that turns the LED on as the first thing in `setup()` it takes 4-5 seconds after the upload completes for the LED to turn on. What causes this delay?
(I'm using a generic L082 board.)

GrumpyOldPizza, 2018-08-03:
Good question. A supported board with the LoRa module typically goes
through calibration of LSI / HSI16 and LSE. But this should not take more
than half a second.

But to be honest I never benchmarked boot time.

tve, 2018-08-03:
OK, I'll dig in a see where the time goes...

The board I use has LoRa but not with a TCXO, so nothing to calibrate with :-). I've been using the stm32l0's with Forth and totally bare-bones (i.e. from datasheet&refman) and I've not used Arduino on ARM, so I'm a bit lost in the sea of #defines and overrides... I like that most of your code is on the bare HW as opposed to layers upon layers of some standard code :-). Do you hang out in some chat room (gitter, slack, ...) where I could ask you some quick questions? I can dig through the code but for some stuff you can probably point me in the right direction off the top of your head.

I've started to add support to build sketches with platform.io and also a Generic-L082CZ variant in my fork https://github.com/tve/ArduinoCore-stm32l0, dunno whether you're interested in a PR.

tve, 2018-08-12:
I believe it had to do with me having `#define STM32L0_CONFIG_LSECLK             32768` but not having an external crystal...

GrumpyOldPizza, 2018-08-12:
The code is not really working without LSE. If you don't have that
populated, you might as well stop and rework your hardware.

GrumpyOldPizza, 2018-08-13:
> Can you elaborate? Are you talking about the RTC code or which portions?

RTC and LPTIM use essentially 32768Hz. RTC is used for pretty much all
timing related facilities. It requires a synchronous divider of 2048. That
is simply not doable with LSI. Even worse, LSI is nominal 37kHz, but could
be anywhere between 26kHz and 56kHz.

tve, 2018-08-13:
I see, thanks for the info.

closed 2018-08-13 by GrumpyOldPizza