GrumpyOldPizza/ArduinoCore-stm32l0 #29: how to set the clock speed

tve, 2018-08-03:
How do I set the desired clock speed for a sketch? Is there a way to dynamically change it?

GrumpyOldPizza, 2018-08-03:
It is statically configured throu the Arduino IDE.

tve, 2018-08-03:
Ugh. I'm using platform.io. Can't stand the IDE... DO you happen to know how it gets passed in? Else I'll have to make sense out of all the stm32l0_system.c clock code...

tve, 2018-08-03:
Ok, found it: _SYSTEM_CORE_CLOCK_
Looks like I need to dig through the platform.txt and board.txt stuff to extract the configuration options in general...

closed 2018-08-03 by tve

GrumpyOldPizza, 2018-10-30:
You have to hold BOOT while toggeling RESET to put the chip again into bootloader mode, if you use lower clocks. 

delays and sleep times do not have to be changed.