GrumpyOldPizza/ArduinoCore-stm32l0 #33: Use of TimerMillis seems to interfere with LoRaWAN join

MrSurly, 2018-08-22:
I have a TimerMillis that I'm using to blink LEDs -- it triggers every 100ms.  If I do _not_ activate this timer, the the join seems to occur immediately.  When it's in use, join can take several minutes.

Having the timer > 1000 ms helps considerably.

Is there some sort of timer contention on the STM32L0?

GrumpyOldPizza, 2018-08-22:
There should not be any conention.

Mind sharing the sketch here, so that I can take a peek ?

MrSurly, 2018-08-22:
It's ... very large.  I'll try to make a minimum sketch that will re-create.

GrumpyOldPizza, 2018-08-22:
Oki. What board is that on ?

MrSurly, 2018-08-23:
It's a custom board running the CMWX1ZZABZ-078.  Variant is B-L072Z-LRWAN1. Actual MCU is STM32L082.

Note that I have not nailed it down to any root cause.  I _do_ notice that when I disable the LED timer, then it often connects immediately after the first (sometimes second) join request.  Before, the wait was 5-6 join requests.  

The physical gateway is 5 feet away, and we're using TTN.

GrumpyOldPizza, 2018-08-30:
Is there a quick repro I could take a peek at ?

MrSurly, 2018-09-04:
Yes, sorry, I've been out-of-town since Wednesday.

MrSurly, 2018-09-17:
I'm going to close this for now.  I think TTN is just slow at responding sometimes.

closed 2018-09-17 by MrSurly