GrumpyOldPizza/ArduinoCore-stm32l0 #35: Several channels appear on one up-link

rossbug, 2018-09-13:
Hi
First, very appreciate this great job(Lorawan library). 
My code is shown below. I am using ABP.
void setup( void )
{
LoRaWAN.begin(EU868);
for (unsigned int i = 3; i <16; i++)
{
LoRaWAN.disableChannel(i);
}
LoRaWAN.setADR(false);
LoRaWAN.setDataRate(0);
LoRaWAN.joinABP(appEui, appKey, devEui);
while(!LoRaWAN.joined()) { };
}
void loop( void )
{
if (!LoRaWAN.busy() && LoRaWAN.joined())
{
LoRaWAN.beginPacket(25);
//-------------------
use LoRaWAN.write() to write 9 data bytes
//-------------------
LoRaWAN.endPacket();
}
STM32L0.stop(120000);
}
And i found a very strange thing on the TTN gateway as below picture.
[image]
At the same time, not only one channel but also another channel with 867.7 or 867.5 MHz.
................does it the frequency hoping???
Thank you very much
Ross

GrumpyOldPizza, 2018-09-13:
Yes, it does do frequency hopping. You disabled all but 3 channels, so yes, it will be using them.

closed 2018-09-13 by GrumpyOldPizza

rossbug, 2018-09-13:
Very thank you for response.

Sorry for my poor knowledge about lorawan.

Does this frequency hopping is added in Lorawan V1.0.2.

Because this result is not appear in RN2483 which is only V1.0.1 or just RN2483 does not enable this function.

Thank you very much

Ross

________________________________
寄件者: Thomas Roell <notifications@github.com>
寄件日期: 2018年9月13日 下午 08:12:49
收件者: GrumpyOldPizza/ArduinoCore-stm32l0
副本: 陳 建至; Author
主旨: Re: [GrumpyOldPizza/ArduinoCore-stm32l0] Several channels appear on one up-link (#35)

Yes, it does do frequency hopping. You disabled all but 3 channels, so yes, it will be using them.

rossbug, 2018-09-28:
Dear Thomas Roell

Very thank you for the support on this project.

I have a question on the LoRaWAN.setAntennaGain(2.0)

What is the unit of 2.0.

Because the TX current is too high(over 120mA) when I do this setting

LoRaWAN.setAntennaGain(2.0)

LoRaWAN.setTxPower(14)

I want to close the PA_boost and use RFO_HF pin to deliver 14dbm power.

What should i do.

Additionally, I want to do LoRaWAN certification.

It seems this library has already embed the TestMode for certification of LoRaWAN 1.0.2 specification.

(Very impressive work)

I want to confirm the version of certification is LoRaWAN 1.0.2 specification, right?

And my code is shown below (OTAA version), does anything need to modify for running the TestMode ?

Thank you very much

Ross

void joinCallback(void)
{
    if (!LoRaWAN.joined())
    {
          LoRaWAN.rejoinOTAA();
    }
}

void setup( void )
{
    LoRaWAN.begin(EU868);
    LoRaWAN.setComplianceTest(true);

    LoRaWAN.addChannel(0, 868100000, 0, 5);
    LoRaWAN.addChannel(1, 868300000, 0, 5);
    LoRaWAN.addChannel(2, 868500000, 0, 5);
    for(unsigned int i=3; i<16; i++)
    {
      LoRaWAN.disableChannel(i);
    }
    LoRaWAN.setADR(true);
    LoRaWAN.setTxPower(14);
    LoRaWAN.setRX2Channel(869525000, 3);
    LoRaWAN.setDutyCycle(false);

    LoRaWAN.onJoin(joinCallback);

    LoRaWAN.joinOTAA(appEuiNew, appKey, deviceEui);

    while(!LoRaWAN.joined())
    {
      STM32L0.stop(500);
    }

}

void loop( void )
{
    if (LoRaWAN.joined() && !LoRaWAN.busy())
    {
        LoRaWAN.beginPacket(2);
        LoRaWAN.write(0xef);
        LoRaWAN.write(0xbe);
        LoRaWAN.endPacket();
    }
    STM32L0.stop(5000);
}

________________________________
寄件者: Thomas Roell <notifications@github.com>
寄件日期: 2018年9月13日 下午 09:24:33
收件者: GrumpyOldPizza/ArduinoCore-stm32l0
副本: 陳 建至; Author
主旨: Re: [GrumpyOldPizza/ArduinoCore-stm32l0] Several channels appear on one up-link (#35)

Frequency hopping has been always part of LoRaWAN.

GrumpyOldPizza, 2018-09-28:
> LoRaWAN.setAntennaGain(2.0)

dBm.

> LoRaWAN.setTxPower(14)
> I want to close the PA_boost and use RFO_HF pin to deliver 14dbm power.

Internally PA_BOOST is automatically selected if the selected TxPower is > 15dBm. However the internal selected power depends upon the MaxEIRP for your region as well as the antenna gain ...

phyTxPower = ( int8_t )floor( ( maxEirp - ( txPowerIndex * 2U ) ) - antennaGain );

For EU868 the maxEirp is 16dBm, your selected txPowerIndex would be 1 (well you selected 14dbm). Considering the antenna gain, the radio drivers sees 12dBm, which is routed via RFO.

> It seems this library has already include the TestMode for certification.

Yes, but that is still work in progress. Please contact me offline via grumpyoldpizza@gmail.com

> I want to confirm the version of certification is LoRaWAN 1.0.2 specification, right?

1.0.2B.

> And my code is shown below (OTAA version), does anything need to modify for running the TestMode ?

LoRaWAN.setComplianceTest(true);

rossbug, 2018-10-01:
Dear Thomas Roell

I have send my product to LoRaWAN certification.

And we (me and Authorised Test House) found some problem as below.

1. After the device enter the test mode, the TX never send any packet.

And i delete the if (!_Joined) and if (_tx_busy) in beginPacket, the TX start sends packet.

Does there another solution?

2. The _ComplianceTestCount does not increment.

3. In the send(), there is a lot of parameters does not have LoRaWAN.

Ex.   _ComplianceTestConfirmed  change to LoRaWAN._ComplianceTestConfirmed

We found this problem because the 0x02 and 0x03 (Confirmed and Unconfirmed) command is fail.

4. 0x06 command is fail, i can not explain exactly why but just suggest the code as below

    case 6: // Disable TestMode and revert back to normal operation (ix)
          stm32l0_system_reset();
    break;
Just rest the MCU and it can pass the certification. (For OTAA sample)

Thank you very much

Ross

________________________________
寄件者: 陳 建至
寄件日期: 2018年9月28日 下午 10:21:10
收件者: GrumpyOldPizza/ArduinoCore-stm32l0; GrumpyOldPizza/ArduinoCore-stm32l0
副本: Author
主旨: RE: [GrumpyOldPizza/ArduinoCore-stm32l0] Several channels appear on one up-link (#35)

Dear Thomas Roell

Very thank you for the support on this project.

I have a question on the LoRaWAN.setAntennaGain(2.0)

What is the unit of 2.0.

Because the TX current is too high(over 120mA) when I do this setting

LoRaWAN.setAntennaGain(2.0)

LoRaWAN.setTxPower(14)

I want to close the PA_boost and use RFO_HF pin to deliver 14dbm power.

What should i do.

Additionally, I want to do LoRaWAN certification.

It seems this library has already embed the TestMode for certification of LoRaWAN 1.0.2 specification.

(Very impressive work)

I want to confirm the version of certification is LoRaWAN 1.0.2 specification, right?

And my code is shown below (OTAA version), does anything need to modify for running the TestMode ?

Thank you very much

Ross

void joinCallback(void)
{
    if (!LoRaWAN.joined())
    {
          LoRaWAN.rejoinOTAA();
    }
}

void setup( void )
{
    LoRaWAN.begin(EU868);
    LoRaWAN.setComplianceTest(true);

    LoRaWAN.addChannel(0, 868100000, 0, 5);
    LoRaWAN.addChannel(1, 868300000, 0, 5);
    LoRaWAN.addChannel(2, 868500000, 0, 5);
    for(unsigned int i=3; i<16; i++)
    {
      LoRaWAN.disableChannel(i);
    }
    LoRaWAN.setADR(true);
    LoRaWAN.setTxPower(14);
    LoRaWAN.setRX2Channel(869525000, 3);
    LoRaWAN.setDutyCycle(false);

    LoRaWAN.onJoin(joinCallback);

    LoRaWAN.joinOTAA(appEuiNew, appKey, deviceEui);

    while(!LoRaWAN.joined())
    {
      STM32L0.stop(500);
    }

}

void loop( void )
{
    if (LoRaWAN.joined() && !LoRaWAN.busy())
    {
        LoRaWAN.beginPacket(2);
        LoRaWAN.write(0xef);
        LoRaWAN.write(0xbe);
        LoRaWAN.endPacket();
    }
    STM32L0.stop(5000);
}

________________________________
寄件者: Thomas Roell <notifications@github.com>
寄件日期: 2018年9月13日 下午 09:24:33
收件者: GrumpyOldPizza/ArduinoCore-stm32l0
副本: 陳 建至; Author
主旨: Re: [GrumpyOldPizza/ArduinoCore-stm32l0] Several channels appear on one up-link (#35)

Frequency hopping has been always part of LoRaWAN.

GrumpyOldPizza, 2018-10-01:
Ross, can you contact me directly via grumpyoldpizza@gmail.com to discuss this.