GrumpyOldPizza/ArduinoCore-stm32l0 #36: [core] UART 'Callback' has not been declared

SloMusti, 2018-09-22:
The following library fails to compile for this core but works well on others (AVR, ESP8266) although it appears to use only standard Wire and Serial functions:  https://github.com/Ralim/BQ34Z100

Traceback below shows that: 

> Uart.h:69:52: error: 'Callback' has not been declared

repeated a number of times. I have not started digging into details on this to know more. TO reproduce, install library and run the provided example.

```
In file included from C:\Users\Uporabnik\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\variants\B-L072Z-LRWAN1/variant.h:48:0,

                 from C:\Users\Uporabnik\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\Wire\src/Wire.h:24,

                 from C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.h:4,

                 from C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp:13:

C:\Users\Uporabnik\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\cores\arduino/Uart.h:69:52: error: 'Callback' has not been declared

     bool write(const uint8_t *buffer, size_t size, Callback callback);

                                                    ^

C:\Users\Uporabnik\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\cores\arduino/Uart.h:82:20: error: 'Callback' has not been declared

     void onReceive(Callback callback);

                    ^

C:\Users\Uporabnik\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\cores\arduino/Uart.h:101:5: error: 'Callback' does not name a type

     Callback _completionCallback;

     ^

C:\Users\Uporabnik\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\cores\arduino/Uart.h:102:5: error: 'Callback' does not name a type

     Callback _receiveCallback;

     ^

In file included from C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.h:4:0,

                 from C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp:13:

C:\Users\Uporabnik\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\Wire\src/Wire.h:112:5: error: 'Callback' does not name a type

     Callback _callback;

     ^

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp: In member function 'int bq34z100::setup(uint8_t, uint8_t, uint16_t, uint16_t, uint16_t)':

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp:155:1: warning: no return statement in function returning non-void [-Wreturn-type]

 }

 ^

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp: At global scope:

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp:127:5: warning: unused parameter 'BatteryChemistry' [-Wunused-parameter]

 int bq34z100::setup(uint8_t BatteryChemistry,uint8_t SeriesCells,uint16_t CellCapacity,uint16_t PackCurrentVoltage,uint16_t AppliedCurrent)

     ^

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp: In member function 'bool bq34z100::readFlash(uint16_t, uint8_t)':

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp:220:10: warning: unused variable 'changed' [-Wunused-variable]

     bool changed = false;

          ^

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp: In member function 'void bq34z100::writeTable(uint16_t, uint8_t)':

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp:252:10: warning: unused variable 'changed' [-Wunused-variable]

     bool changed = false;

          ^

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp: At global scope:

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp:265:6: warning: unused parameter 'subclass' [-Wunused-parameter]

 void bq34z100::checkSum(uint16_t subclass, uint8_t offset)

      ^

C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100\bq34z100.cpp:265:6: warning: unused parameter 'offset' [-Wunused-parameter]

Using library Wire at version 1.0 in folder: C:\Users\Uporabnik\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.7\libraries\Wire 
Using library bq34z100 in folder: C:\Users\Uporabnik\Documents\Arduino\libraries\bq34z100 (legacy)
exit status 1
Error compiling for board B-L072Z-LRWAN1.

```

GrumpyOldPizza, 2018-09-22:
Zero clue. "Callback.h" is included from "Arduino.h". It's compiled/included from a C++ file properly.

GrumpyOldPizza, 2018-09-23:
Downloaded repository, and compiled examples/setup/setup.ino without a hitch. No source code changes.

Closing out.

closed 2018-09-23 by GrumpyOldPizza

SloMusti, 2018-10-06:
This issue is only present on the Ubuntu and not on Windows, what causes it is if Wire.h is included in a library in .cpp file. Simply moving that include to the .h file of the library resolves the problem.

No action is required or this, leaving it here if someone else runs into the issue.

GrumpyOldPizza, 2018-10-06:
That one is odd. Hit another instance of this. Again a *.cpp file that is part of another library includes Wire.h directly without Arduino.h. Looks like in this case the IDE does not prepend Arduino.h (which it does for any *.ino file). 

I could patch this half way by having "Callback.h" included throu "WVariant.h". But that does not seem to be right. Or simply include "Callback.h" for every library that uses it (and add then stm32l*.h and the armv6m*.h files throu "WVariant.h".