GrumpyOldPizza/ArduinoCore-stm32l0 #37: LoRaWAN Class B and Class C support

mhornjak, 2018-09-22:
Hi there, 
great work. Since this is based on the Semtech Lora-Net implementation, is Class B or at least Class C supported? As far as i have seen in the commit comments, i think not - or am i mistaken?
Are there any plans to implenent this in the near future?

Thanks a lot!

GrumpyOldPizza, 2018-09-22:
The official release version of the LoRaMac-node stack does not support
Class B, or LoRaWAN 1.0.3. Class C without the multicast updates in 1.0.3
is not that useful.

Shortly after LoRaMac-node 4.4.2 is released Class B and Class C support
will be added to the LoRaWAN class.

jcwren, 2019-10-31:
The latest versions of the LoRaMAC stack now support class B (and I think C, but I haven't paid attention to that). I've done a Class B product using the ST LoRaWAN stack and it mostly works. 

The harder part is finding a back-end that supports Class B.  The Things Network does not yet. MachineQ does, but that's pay-to-play. SemtechIO might, but their site is painful.

You can spin up your own using loraserver.io, which is what I've done. I went with the the containerized version. Biggest issue I've had is that on some occasions I've seen loraserver.io appear to be transmitting on the wrong beacon channel. The newest version may not have this issue, but it hasn't been running long enough to tell for sure.