GrumpyOldPizza/ArduinoCore-stm32l0 #38: UART version of SerialX.onReceive does not wake up from Stop mode

romansoft, 2018-09-24:
SerialX.onReceive is supposed to generate an interrupt when there is activity on serial interface. However, when the processor is in Stop mode, it does not wake up upon UART activity. In Active mode, the interrupt works correctly. USB version works correctly in both Stop and Active modes.

Here is the test case for Cricket board:

```
#include <STM32L0.h>

void dummy(){
  STM32L0.wakeup();
}

void setup() {
  pinMode(PIN_LED, OUTPUT);
  digitalWrite(PIN_LED, HIGH);
 
  Serial2.begin(57600); // UART 2
  Serial2.onReceive(dummy); // pin A3 should trigger interrupt, but doesn't in Stop mode
  
  Serial.begin(57600); // USB
  Serial.onReceive(dummy); // USB activity triggers interrupt
}

void loop() {

  digitalWrite(PIN_LED, HIGH);
  delay(500);
  digitalWrite(PIN_LED, LOW);
  delay(200);
  
  STM32L0.stop();        // Enter STOP mode and wait for an interrupt
}

```

A3 triggers interrupt (ie. led blinks) the first time after reset (processor in Active mode), but not after going into Stop mode.

GrumpyOldPizza, 2018-09-24:
You need to enable the wakeup from STOP. Currently this is done via Serial.setWakeup(). The API for this might change though (I am currently working on getting all this wakeup logic more consistent).

Please remember that above 38400 UART wakeup from STOP requires HSI16 to be active in STOP mode. So you pay an extra 100uA for that.

There is internally another mode, where simply the RX line is used as a regular INT line on entering STOP mode. So this also needs to be exposed properly.

Ah lastly, the onReceive() callback fires only once when new data arrived after your last read() operation. So you cannot use it to track activity via LED as in your example.

romansoft, 2018-09-24:
Thanks for quick reply and the insights. Using Serial.setWakeup() was not apparent. I tried setting this, but it still doesn't wake up from Stop. Same behavior as before. 
```
  Serial2.begin(9600); // UART 2
  Serial2.setWakeup(true);
  Serial2.onReceive(dummy); // pin A3 should trigger interrupt, but doesn't in Stop mode

```
The example is pretty synthetic, using LED to see if the interrupt and wake up events happen.

romansoft, 2018-09-25:
I tested again (on B-L072Z-LRWAN1 this time), and the wakeup from Stop worked just fine with the added Serial2.setWakeup(true). Not sure why Cricket was not obeying before. Perhaps I has a loose connector on the pin header. 
I think we can close the issue.

GrumpyOldPizza, 2018-09-25:
I am pretty sure Cricket works just fine. The GNSS library is using that
quite a bit.

romansoft, 2018-09-25:
Yes, I saw that. It's just GNSS was using UART1 and I was using UART2 on Cricket, so I wasn't sure if there was any UART instance limitation. But, I verified on B-L072Z-LRWAN1 that UART2 works as expected. I have no reason to suspect any issue with Cricket board, since UART2 works fine, in general.

closed 2018-09-25 by GrumpyOldPizza