GrumpyOldPizza/ArduinoCore-stm32l0 #40: OTAA join not working after session save

romansoft, 2018-09-27:
Not sure if this is intentional behavior or not, but, after I perform setSaveSession(true) and joint the network the first time, subsequent joinOTAA's have no effect, ie. no MAC commands are sent to a network server and no onJoin event is triggered. It looks that uplinks continue with the previous frame count value. 
So, in effect, joinOTAA acts like rejoinOTAA which is unexpected and undesirable.

The reason I'd like joinOTAA to start a fresh join process with the server is allow for recovery from potential state corruption in either a node or in a network server without waiting very long time for the MAC stack to determine no gateways are connected. Additionally, on the system reboot, I perform join and wait for onJoin event before proceeding further. If it never happens, it causes some issues.

GrumpyOldPizza, 2018-09-27:
It is intended behavoir.

rejoinOTAA() is intended as forced rejoin after a connection loss (by ADR detection or confirmed message loss detection).

So a better way would be to junk "setSaveSession()" and simply set "joined()" to true directly after LoRaWAN.begin() if the last session is still active (OTAA or ABP), i.e. before you force a joinOTAA()/rejoinOTAA() ?

The whole purpose of the EEPROM there is to allow a continue of a OTAA session across reset boundaries. 

One thing you have to consider is that with "devNonce" being only 16 bit, and each "devNonce" only being usable once, you can only do a joinOTAA() 65536 times per appEUI ...

romansoft, 2018-09-27:
OK, rejoinOTAA works as I expect, it allows to reestablish the last session. This is very useful in some scenarios, for example, after replacing a battery. 
However, there are situations, where the shared session between node and NS may no longer be valid. For example, state corruption on either side (like frame counter reset due to a glitch) or a node being moved under service of another gateway with a different subband (most public network operators require a node to detect link loss and reacquire the new channel plan) or when DevEUI/AppEUI/AppKey were modified to join a different network operator. These devices may be placed untouched in inaccessible locations over very long period of time while the network around them evolves, so they need to be robust. 
In those situations, the only logical option is to force the join process with the NS to reestablish a new session. Otherwise, the node may become stuck in unconnected state and not able to recover. My expectation was that joinOTAA() does just that - triggers the node to start a new session, while rejoinOTAA() just restores the current session. (If joinOTAA() behaves like rejoinOTAA(), why even have rejoinOTAA() function?)
So, how do I force a new session, ie. clear current session state settings in eeprom while keeping initial credentials (AppKey, AppEUI)?
There are also situations when the node needs to be restored to the factory state, say after load testing or servicing. How to do a "factory reset", ie. clear the whole eeprom state - session and credentials?

romansoft, 2018-09-27:
Thanks for the clarifications. I had a wrong understanding of rejoinOTAA() behavior and purpose. I thought it restored the last session, same as joinOTAA does. 
Anyway, I understand your dilemma about the API definition. My thinking is that there are only a few well defined scenarios when you want to restore the existing session (eg. power loss), but many more not well defined scenarios when you want to trigger a new join (eg. unforeseen errors, user pushing a reset, etc.). So, I would make new join trigger as default behavior and the session restore something the application has to request specifically. Issue recovery is more critical to me than power optimization. But, it's my two cents.

I tried setSaveSession(false) yesterday, but it did not seem to clear the session settings. Uplinks continued as before, after joinOTAA() timed out. Is there a long latency to when eeprom clearing takes effect after this command or should eeprom clear immediately? I'll try again later.

romansoft, 2018-09-27:
True, network carriers hate frequent rejoins as it reduces gateway's capacity. But, I don't see rejoin events frequent on production devices (perhaps few times a year), and definitely much less than 65536 times over the lifetime of a node (unless it's a defective node or botched firmware). For development devices, it's a different matter.
I guess the default behavior you propose is OK to protect uninformed developers from stressing the network. As long as you provide the restore flag, the advanced developer can choose the appropriate behavior for their use case.

`The EEPROM writes are deferred, but the in SRAM copy is current all the time ...`

This explains what I saw. I was doing device reset 200ms after setSaveSession(false). SRAM would be cleared, but EEPROM might've not.  When does EEPROM write take place?

GrumpyOldPizza, 2018-09-27:
EEPROM takes 6.4ms for a 32bit write. So to write back the full 256 bytes (or whatever it is) might take a while.

GrumpyOldPizza, 2018-09-28:
Ok, my head is spinning. You want an override for not restoring the previous session. 

That is trivial, just don't use setSaveSession(true) before you do do a joinOTAA(). 

I agree that the API could be modified to use a different more intuitive scheme. But the functionality is there as is.

closed 2018-09-28 by GrumpyOldPizza

romansoft, 2018-09-28:
@GrumpyOldPizza, not so trivial. Please consider the scenario when Murata module is used as modem and the application firmware is on another MCU. For security reasons, it is desired to keep the lora credentials out of application code, programmed only once into the Murata module. Correct me if I'm wrong, but since there there is only one method of saving the state, which saves both initial credentials and session state, it would not be possible to achieve this objective. 

I tested setSaveSession(false) again, waiting more than sufficient time (2 sec) for the EEPROM write to complete, but it still does not seem to work. Issuing an uplink command afterwards, performs uplink transmission and returns without any error. I'd expect that if the session credentials are wiped out, communication should stop working.

GrumpyOldPizza, 2018-09-28:
The credentials are not part of the session save process. You call
::setAppEui() and it gets stored in EEPROM. Then you can call ::joinOTAA()
without arguments and it uses the commissioning from EEPROM. The set in
setSaveSession() is more along the line of session keys, and ADR settings
and channels.

GrumpyOldPizza, 2018-09-30:
Still toying with that. I *think* I'll change the scheme there, so that a session is always saved, and restored post reset (at ::begin() time). Hence ::joined() will be true, and you simply can continue your session.

If you want to join another session (gateway), you can simply call joinOTAA() / joinABP().

Not all is clear to me, but that would mean that some parameters would apply only to the next join (like channel mask/frequencies), or rejoin. Thus they would not affect the current session. So the parameters need to be split (into a static and a dynamic set) and saved/restored differently.