GrumpyOldPizza/ArduinoCore-stm32l0 #42: DigitalWrite issue on pins >13 number

SloMusti, 2018-09-29:
I have been testing the board variant Grasshopper, however not on original hardware but on custom boards and STM development boards.

Checking the variant file, all seems in order, however digital write does not work on any pin greater then 13, namely analog pins and extra pins. I am quite sure @GrumpyOldPizza implemented this well and thus kinds ask someone else to test this.

Variant pin definition:
https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/e2212a4162e55f76009a97756b065a7a7232793b/variants/Grasshopper-L082CZ/variant.cpp#L67

GrumpyOldPizza, 2018-10-01:
Is there an issue with Grasshopper itself ? If so, what exactly does not work ?

SloMusti, 2018-10-01:
I was unable to get any digital pin to work with number greater then 13, so doing the digital write at Analog pins and on extended pins, such as PA11 and PA12. Can you confirm this is working for you and only I am facing this issue on multiple boards?

GrumpyOldPizza, 2018-10-01:
Some pins are only mapped as reference in the variant.cpp file. Those you cannot read or write. But their mapping is used internally. PA11/PA12 are used for USB, hence you cannot read/write them.

The analog pins are not any different than the regular pins. However the first analog pin is index 16, so you might be off by 2.

SloMusti, 2018-10-01:
PA11 and PA12 would be great for standalone use as on some boards without USB this gives extra GPIO.

GrumpyOldPizza, 2018-10-01:
It seems that what you want is to write your own variant file.

Does the digital IO on the "analog" pins work now that the "off by 2" had been identified ?

SloMusti, 2018-10-02:
Correct, I have tested everything and it works now, have started preparing my own variant file and will share that shortly.

closed 2018-10-02 by SloMusti

GrumpyOldPizza, 2018-10-02:
Word of caution. I will not include any variants for boards I have no access to.

You might want to consider forking off ArduinoCore-stml32l0 and maintain your own variant, removing all others.