GrumpyOldPizza/ArduinoCore-stm32l0 #43: LoRaWAN.confirmed() returns false even though packet shows confirmed through TTN

christopherdro, 2018-10-02:
I double checked that the issue wasn't specific to my code by running this example here. https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/master/libraries/LoRaWAN/examples/LoRaWAN_Confirmed/LoRaWAN_Confirmed.ino

Here is a screenshot of the confirmation from TTN's console.
[image]

My end goal here is to write the un-confirmed packets to flash and batch upload them once the sensor comes back within range of a gateway.

GrumpyOldPizza, 2018-10-02:
Is there a way to track which RX slot is used to send the confirm ? I
retested the code here locally with my gateway (US915 in that case), and it
comes back fine.

christopherdro, 2018-10-04:
I ended up testing on a few different gateways only to find out that the issue was actually specific to my sketch code. 😟 

When I was referencing the example file in my code I excluded the empty while loop [here](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/master/libraries/LoRaWAN/examples/LoRaWAN_Confirmed/LoRaWAN_Confirmed.ino#L85-L87). Strangely, I was can reproduce the issue by removing this while loop in the example.

It was a bit hard to track down but it looks like a delay of at least 2 seconds is needed after sending the packet to allow time for the rx windows.
 
Is it possible that the execution of the while loop conditional takes long enough to have received the downlink packet? 

In the end I was able to resolve the issue by adding `delay(2000)` after calling `sendPacket`.

GrumpyOldPizza, 2018-10-04:
Ok, there is a fundamental misunderstanding how the code works. When you issue a LoRaWAN.endPacket(), then the system is starting the TX/RX1/RX2 sequence in the background (possibly deferred if duty cycling is applied). During that time LoRaWAN.busy() returns true. Obviously the status of whether a ACK was received or not is only valid after TX/RX1/RX2 is done, i.e. LoRaWAN.busy() returns false. 

This is what the original code in the examples does.

Adding a "delay(2000)" does not really help. You need to make sure that LoRaWAN.busy() returns false.

You might be sneaky and save power doing something like this:

while (LoRaWAN.busy()) { __WFE(); }

closed 2018-10-04 by GrumpyOldPizza

christopherdro, 2018-10-04:
I check to make sure LoRaWAN.busy() returns at the start of the loop. Here is a reference to it.