GrumpyOldPizza/ArduinoCore-stm32l0 #46: hex file generate

ucdos123456, 2018-10-17:
hi, i found i cannot generate hex file. and i try to modify platform.txt to add`## Create output (.hex file)
recipe.objcopy.hex.pattern="{compiler.path}{compiler.objcopy.cmd}" {compiler.elf2hex.flags} {compiler.elf2hex.extra_flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.hex"

## Save hex
recipe.output.tmp_file={build.project_name}.hex
recipe.output.save_file={build.project_name}.{build.variant}.hex` 

but compile error! Can you have a look about this?

GrumpyOldPizza, 2018-10-17:
Perhaps using the right flags would hep ? "{compiler.elf2hex.hex.flags}"

Why would you want a *.hex file ? It's not needed for the Arduino IDE.

ucdos123456, 2018-10-17:
I have to download the program to a board by hex method.

GrumpyOldPizza, 2018-10-17:
Ok, so the issue is not relevant to this Arduino Core.

closed 2018-10-17 by GrumpyOldPizza

SloMusti, 2019-08-28:
The complete code for adding HEX file output is:

```
## Create output (hex file)
recipe.objcopy.hex.pattern="{compiler.path}{compiler.elf2hex.cmd}" {compiler.elf2hex.hex.flags} {compiler.elf2hex.extra_flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.hex"
recipe.output.tmp_file={build.project_name}.hex
recipe.output.save_file={build.project_name}.{build.variant}.hex
```

and needs to be added to: https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/master/platform.txt#L99