GrumpyOldPizza/ArduinoCore-stm32l0 #56: TLera Long Cricket

MartinNel, 2019-01-14:
Can you provide an entry in the boards.txt file for the new Long Cricket from Tlere Corp?
Eg:  XXXXX.build.board=STM32L0_LONG_CRICKET

We try to get our code to be source code compatible for most of the Tlere Corp devices and it would help if we could distinguish this board from the other boards using a #ifdef like for the other boards (eg ARDUINO_STM32L0_CICADA, ARDUINO_STM32L0_CRICKET and ARDUINO_STM32L0_GNAT)

GrumpyOldPizza, 2019-01-14:
Currently "Long Cricket" and "Cricket" are both handled by the same "Cricket" variant. The code internally figures out whether it's a BMA280 or BMA400 on board to get the different mapping done.

I see why you'd want a separate variant thought.

MartinNel, 2019-01-15:
For example:

`  #if defined ARDUINO_STM32L0_CICADA || defined ARDUINO_STM32L0_CRICKET

    // Sampling
    float Vbat = STM32L0.getVBAT();     

  #elif defined ARDUINO_STM32L0_GNAT || defined ARDUINO_STM32L0_LONG_CRICKET

    // Opens FETs during sampling 
    digitalWrite(PIN_BAT_MEAS_CTRL,HIGH);
    float Vbat = 1.27f * STM32L0.getVDDA() * ((float) analogRead(PIN_BAT_DIVIDER)) / 1024.0f;
    digitalWrite(PIN_BAT_MEAS_CTRL,LOW);
         
  #endif `

GrumpyOldPizza, 2019-01-15:
The why is understood. I will add a variant.