GrumpyOldPizza/ArduinoCore-stm32l0 #59: IIC slaver can only receive 27 bytes

ucdos123456, 2019-01-16:
every time, if slaver receive 27 bytes from master, it cannot receive anymore. i'm not sure why?

GrumpyOldPizza, 2019-01-16:
The limit should be 32, not 27. Have you tried to increase the default
buffer size ?

GrumpyOldPizza, 2019-01-17:
What is supposed to happen is that after 32 bytes STM32L0 is supposed to stop receiving and NACK every subsequent byte till a STOP condition is received. It should not crash or do anything bad.

The reason for the "no refill" is that if you do that, you don't know when a stop condition showed up on the bus (which you may need for tracking purposes). ArduinoCore-samd for example does the same thing for the same reason.

What is missing is a way to properly size the buffers other than by editing Wire.h.

ucdos123456, 2019-01-18:
Thx for your reply,

The test i did is : 
Sending ONE byte several time from my master (this master managed to communicate with several i2c devices with no issue like a PCA9674), which means :
-start condition | address +w | One byte | stop condition
-wait 1 second and do it again 32 times.
after 32 bytes sent and receive and read no more bytes accepted and unresponding i2c bus.

I am pretty sure that rxBufferIndex is never reset to 0 at any time, i did spend more time on the library to be sure at 100%.
The change only concern top level of the library the protocol in itself is not impacted, you still know when you start receiving and stop.

GrumpyOldPizza, 2019-01-31:
There is a new Wire library version coming that might fix that.

I am concerned about the fact that your comment refer variables and code that not part of ArduinoCore-stm32l0.

GrumpyOldPizza, 2019-02-04:
New code uploaded. 0.0.10.

closed 2019-02-04 by GrumpyOldPizza