GrumpyOldPizza/ArduinoCore-stm32l0 #61: Downlink disable

SloMusti, 2019-01-17:
Just wondering if there are any options for downlink disable in order to be able to save power.  For example I know I can change the wait times and offsets like this, but diasble is nto performed in this case.

```
  LoRaWAN.setReceiveDelay(0);
  LoRaWAN.setRX1DrOffset(0);
```

Use-case scenario is to be able to disable downlinks and rx windows either in all cases and breaking direct LoraWan compatibility or disabling it when not expecting a downlink.

GrumpyOldPizza, 2019-01-19:
Something like:

`
    LoRaWAN.setDownLink(bool onoff);
`

No, that is not supported. Should I add that ? Is control per RX1/RX2 needed ?

SloMusti, 2019-01-19:
That would work, there is no need for separate RX1/RX2 control I believe as this functionality is really for purposes of going to sleep as soon as possible after transmission. This change would need to be implemented in a way, that you can use it prior to every transmission.

GrumpyOldPizza, 2020-05-06:
Need to add a few other things to the LoRaWAN class. 

What about:

LoRaWAN.disableRxWindow1()
LoRaWAN.disableRxWindow2()

GrumpyOldPizza, 2020-05-06:
Added:

LoRaWAN.setRxWindows(bool onoff);

So you want to first join (or continue from your previous join), and then disable the RxWindows(). However ADR will of course not work, LinkCheck will not work ... So not sure this is useful, other than for testing.

closed 2020-05-06 by GrumpyOldPizza