GrumpyOldPizza/ArduinoCore-stm32l0 #62: LoRaWAN.busy() stays true after gateway join but only when device can see multiple gateways

NetSensors, 2019-01-25:


GrumpyOldPizza, 2019-01-25:
Ack ;-)

No idea what that could be and no idea how to debug that. 

In theory the code tries 3 times and then gives up (resetting busy, line 3524, LoRaWAN.cpp). So assuming there is corruption because more than one gateways answers (which should not be the case), it should not matter, because the code would retry, using a new "devNonce", so that a gateway would see a new join attempt. And it seems unlikely that more than one gateway would consistently corrupt the signal back to back.

The code had been tested that it would give up after the right number of attempts. It had been tested that it would do the right thing it the LoRaMac-node stack would return an error.

NetSensors, 2019-02-04:
I have investigated further and this appears to be a gateway issue as we are seeing it with other LoRaWAN modules.

closed 2019-02-04 by NetSensors