GrumpyOldPizza/ArduinoCore-stm32l0 #63: Detecting if USB power (VBUS) is connected.

matt-kiwi, 2019-01-29:
I want to detect when USB power is present, then my software will enter a "configuration mode"
I note that USB_VBUS is physically connected to Pin20/PA8

I figure something like this;
#define PIN_USB_VBUS ??
pinMode( PIN_USB_VBUS, INPUT );
int hasUsbPower = analogRead(PIN_USB_VBUS)>0;

NetSensors, 2019-01-29:
I used the below that I found in one of the examples. It works well  

  float VDDA, VBUS

  VDDA = STM32L0.getVDDA();
  VBUS = STM32L0.getVBUS();

  #ifdef MY_DEBUG
  Serial.print("VDDA = "); Serial.println(VDDA, 2); 
  if(VBUS ==  1)  Serial.println("USB Connected!"); 
  #endif

>

matt-kiwi, 2019-01-30:
I used STM32L0.getVBUS()
But it dosn't seem to update after the board has booted, I remove the USB power after boot, and STM32L0.getVBUS() still returns 1

hallard, 2019-11-30:
I'm facing the same issue today, unable to check when USB is reconnected.
But to be honest it's not a software issue more than a hardware one

I have exactly the same problem with a custom board, it's coming from the reverse forward current of the Schottky diode protecting VUSB line if you're using gnat board like.

When powering on battery (says 3.6V) without usb connected the VBUS line is equal to 3.6V changing the diode with 1N4148 (no forward current) solve the issue meaning the regulator is powered with 4.4V instead of 4.9V, not really an issue.

Kris is aware of this issue detected for another project, I missed the VBUS one also, so my card have also this issue.