GrumpyOldPizza/ArduinoCore-stm32l0 #64: setSaveSession increments uplink counter by 256 on reboot but not on reset

NetSensors, 2019-01-29:
I am using the LoRaWAN_Save_Session.ino from the examples folder. 

The session saves successfully and restores successfully both when power cycled and when reset using the button and when using STM32L0.reset()

However when power cycled the device jumps from its current counter position to 256. If powercycled again it jumps to 512 etc

All other reset methods on the device resume the counter incremented by one as expected. 

I have tried this on a few devices and they all respond the same.

GrumpyOldPizza, 2019-01-29:
Good question. The EEPROM has a endurance of 100k cycles. In tests at room temperature I got about 2000k cycles. Worst case is 2k cycles at 145C. The UpLinkCounter/DownLinkCounter use a wear leveling scheme that increases this endurcance by 32x. So in a nutshell there is a limit how often you can write a UpLinkCounter/DownLinkCounter.

The code puts the lower 8 bits into BACKUP RAM (RTC), and only writes back the upper 24 bits. If you go throu a power cycle, this  BACKUP RAM loses it's content, and hence you have to move the UpLinkCounter forward to the next viable one (and recover the DownLinkCounter at the next downlink). 

I figured a worst case is that if you send a packet every 20 seconds, for 10 years, with only 2k cycles available, then this would still work.

NetSensors, 2019-02-04:
Closing this issue as it is expected behaviour. Reopening issue surrounding LoRaWAN.setSaveSession(true);

closed 2019-02-04 by NetSensors