GrumpyOldPizza/ArduinoCore-stm32l0 #69: Issue using PA12

PAk-CatchFire, 2019-02-15:
Hi there. I am having an issue using a B-L072Z-LRWAN1 board and wanted to use PA12 as an output pin. I have checked variant cpp and added (in the digital block - position 3) the line:

` { GPIOA, STM32L0_GPIO_PIN_MASK(STM32L0_GPIO_PIN_PA12), STM32L0_GPIO_PIN_PA12,           (PIN_ATTR_EXTI),                               PWM_INSTANCE_NONE,  PWM_CHANNEL_NONE, ADC_CHANNEL_NONE  },`

I have alse changed:
`void initVariant()
{
    //CMWX1ZZABZ_Initialize(STM32L0_GPIO_PIN_PA12, STM32L0_GPIO_PIN_PA11);
CMWX1ZZABZ_Initialize(STM32L0_GPIO_PIN_NONE, STM32L0_GPIO_PIN_PA11);

}`

Then in variant.h I have added:
#define PIN_D3               (03ul)
but with no luck.

On the other hand, I have done the same for PA5 (as Analog1 - A1) and it worked fine for that pin.

Is there anything I am missing?
Thank you

GrumpyOldPizza, 2019-02-15:
I might be wrong, but there is SB15, which is not fitted per default. Without that one PA12 is not routed to the headers.

PAk-CatchFire, 2019-02-15:
Thank you for your answer Thom. 
I have tested it with SB15 already fitted and in a custom board I have made, where I use that pin to control a switch, but with no positive results.

Is there any incompatibility as it is a USB pin?
Regards

GrumpyOldPizza, 2019-02-15:
I don't think there is an issue with USB vs. not. Only if you have USB enabled of course.

PAk-CatchFire, 2019-02-15:
Thank you again. 
Would you mind to check if it is working for you?

I have used:
```
/*
 * Pins descriptions
 */
extern const PinDescription g_APinDescription[PINS_COUNT] =
{
    // 0..13 - Digital pins 
    { GPIOA, STM32L0_GPIO_PIN_MASK(STM32L0_GPIO_PIN_PA3),  STM32L0_GPIO_PIN_PA3,            (PIN_ATTR_EXTI),                               PWM_INSTANCE_NONE,  PWM_CHANNEL_NONE, ADC_CHANNEL_NONE },
    { GPIOA, STM32L0_GPIO_PIN_MASK(STM32L0_GPIO_PIN_PA2),  STM32L0_GPIO_PIN_PA2,            (PIN_ATTR_EXTI),                               PWM_INSTANCE_NONE,  PWM_CHANNEL_NONE, ADC_CHANNEL_NONE },
    { GPIOA, STM32L0_GPIO_PIN_MASK(STM32L0_GPIO_PIN_PA10), STM32L0_GPIO_PIN_PA10,           (PIN_ATTR_EXTI),                               PWM_INSTANCE_NONE,  PWM_CHANNEL_NONE, ADC_CHANNEL_NONE },
    { GPIOA, STM32L0_GPIO_PIN_MASK(STM32L0_GPIO_PIN_PA12), STM32L0_GPIO_PIN_PA12,           (PIN_ATTR_EXTI),                               PWM_INSTANCE_NONE,  PWM_CHANNEL_NONE, ADC_CHANNEL_NONE  },
...
```

Thank you

GrumpyOldPizza, 2019-02-15:
I cannot check that here, because my B-L072Z-LRWAN1 is setup for SFOX, so I don't want to mess with PA11/PA12 ...

Mind asking why you want to do that in the first place ?

PAk-CatchFire, 2019-02-18:
Nevermind, it was a bad soldering on that pin. The variant.cpp file is working fine.

Just one last question, is it possible to control the DBG_SX1276_DIOX pins from the code?
Thank you

GrumpyOldPizza, 2019-02-23:
No there is no control over the DBG_SX1276_DIOx pins. They are directly handled by the radio driver internally.

closed 2019-02-23 by GrumpyOldPizza