GrumpyOldPizza/ArduinoCore-stm32l0 #70: Sigfox examples

PAk-CatchFire, 2019-02-23:
Hello, how can I find a Sigfox example for this Core?
Regards

GrumpyOldPizza, 2019-02-23:
There is no Sigfox support in this Core.

NetSensors, 2019-03-21:
There is a sigfox example here but make sure you make the hardware mods mentioned as the TCO_VDD is powered by a different GPIO to the LoRaWAN one. I'm afraid it's not an arduino based example though.

https://github.com/aureleq/muRataSigfox

GrumpyOldPizza, 2019-03-21:
That is incorrect. Other than on GNAT, X-CUBE-SFOX will NOT work. To get this working on GNAT you need to change the different VDD_TCXO pin, as well as the different mapping for DIO4/DIO5. Not that tricky, but painful.

In general SFOX on SX1276 for this module requires a different clocking for APB2 (higher power consumption) and a bunch of other non-trivial changes. Significant issue is the license that comes along with X-CUBE-SFOX, that kind of prevent and integration into ArduinoCore-stm32l0. However should SFOX ask me to do that with a different license ...

closed 2019-04-15 by GrumpyOldPizza