GrumpyOldPizza/ArduinoCore-stm32l0 #72: Using this core in PlatformIO

BoelD, 2019-03-06:
I appreciate working with this core, but I am not a big fan of the Arduino IDE. I prefer PlatformIO in combination with Visual Studio Code. So I wonder if anybody here got this core working in PlatformIO? I have been trying to make it work for some time after I found [this thread](https://community.platformio.org/t/how-to-add-a-different-arduino-core/4698/6), but couldn't get it to work.

BoelD, 2019-03-08:
I came across an alternative solution. Instead of platformIO it's possible to use Arduino for Visual Studio Code and this works together with this core without any problems.

GrumpyOldPizza, 2019-03-08:
Mind putting up a link for me there (howto and such) ? I have to admit that I do develop with a totally different workflow, and I do not use the Arduino IDE a lot. For me the missing debugger support is the biggie. Does "Visual Studio" resolve that ?

BoelD, 2019-03-10:
It's pretty straightforward. Visual Studio can be downloaded [here](https://code.visualstudio.com/). After installing, press` [CTRL]-[SHIFT]-X` and install the Microsoft Arduino for Visual Studio Code extension. And make sure you have the Arduino IDE installed as well, since the extension relies directly on the IDE itself.

After istalling, in Linux I had to manually set the directory of the Arduino IDE in VS Vcode, in `{references>Setings>Extensions>Arduino Configuration>Arduino:Path`. In my case, the path is `/home/david/Programs/arduino-1.8.7` (Linux Ubuntu).
Under Windows, it seems to be dependent on which system you use. On a Windows 8 machine it worked without manually editing the path, on a Windows 10 machine I had to add the path to the Arduino IDE manually.

To edit an Arduino project, open the folder in the VS Code workspace and in the Command Pallette select `Arduino: Initialize`. Then you can select the board (eg Grasshopper) and start working. Keep in in mind, that the board you want to work with, has to be installed in the board manager. You can do this in the Arduino IDE, or from VS Code in` Command Pallette:Arduino:Board Manager`.

If you get include errors, it may be necessary manually add some include paths to c_cpp_properties.json. Under Windows 8 and Linux I had no problems, but on a Windows 10 machine I had to add the libraries directory from the STM32L0 core.

silardg, 2019-12-28:
@BoelD I have everything working but the ST-Link is not showing up in the programmers list?
Did you have issue with this?

BoelD, 2019-12-28:
I just checked and don't see the ST Link either. I always used the USB bootloader for uploading. That doesn't show in the list either, but seems to be selected automatically when selecting the Grasshopper board. Do you really need the ST Link? What if you select the B-L072Z-LRWAN1 board? Uploading via ST Link is the default for this board, does it work with the ST link then?

@GrumpyOldPizza could this be an issue what can be easily solved in your package?

GrumpyOldPizza, 2019-12-28:
It should really just work. You first need to select a programmer, which should be the right one per default. And then you just use the "download via programmer" option.

Yes, there seems to be a typo for grasshopper, but that is easily fixed.

silardg, 2019-12-28:
@BoelD @GrumpyOldPizza I have developed my own board with the cmwx1zzabz, but the wiring is the same as on the B-L072Z-LRWAN1 board. So I'm using an external ST-LINK. With platformio it works, arduino too, but when I edit the project with VSCode and the Arduino module it doesn't. I thought that it will automatically detect the ST-Link and set that, but that is not happening either.