GrumpyOldPizza/ArduinoCore-stm32l0 #73: Support of PA boost on CMWX1ZZABZ

PAk-CatchFire, 2019-03-07:
Hi again.
On Murata's datasheet appears the following:
"Normal output power is +14dBm but a PA boost function can be selected to increase RF output to + 20dBm for long-range applications"

Does this core support that function?
Thank you

GrumpyOldPizza, 2019-03-07:
Yes. Actually RFO goes to +15dBm. PA_BOOST to 20dBm nominally.

PAk-CatchFire, 2019-03-07:
Thank you. How could I enable/disable it?

GrumpyOldPizza, 2019-03-07:
You don't enable/disable that. Depending upon what TxPower you use the
proper path will be selectect.

PAk-CatchFire, 2019-03-10:
Ok, I see....
So could I just call:
` LoRaRadio.setTxPower(14);`

and get PA_BOOST on automatically?

Regards

GrumpyOldPizza, 2019-03-12:
Pretty much. Except up to 15dBm RFO is in use. But if you used 20dBm, then yes PA_BOOST is used.

closed 2019-04-15 by GrumpyOldPizza