GrumpyOldPizza/ArduinoCore-stm32l0 #74: LoRaRadio

SolarDaniel, 2019-03-08:
Hi
First of all,Thanks for your great work.

I'm playing around with the grasshopper module for a longrange remote control.
In my pingpong example with callbacks I use LoRaRadio.receive(1000); to check, if I missed a response from the slave.
I always get a receive callback for the timeout, even when a valid data packet was received.
So I get a receive callback for the data and after the timeout I get a receive callback with no data.
My expectation would be, that when a call to LoRaRadio.receive(1000); ends in receiving data within the timeout period, the timeout timer would be cleared and no timeout callback would be raised.

Furthermore, would it be possible to have an additional callback for the receive timeout instead of calling the receive callback?

Thanks
Daniel

GrumpyOldPizza, 2019-03-08:
The radio in the case you are describing is in continuous receive mode. Hence if you receive a packet you have to stop the radio by putting it either into sleep() or standby() if that is what you want to do.

An additional callback should not be required ... ::busy() will tell you whether this radio is still busy within the the onReceive() callback or not. If you are not busy, then it's a timeout, if you are busy, its a packet.

SolarDaniel, 2019-03-08:
Thank you for the quick response.
Well,I understand. I think I need to learn some more on the LoRa radio.
It's everything there I need for my project,just need to learn how it works.

You spent a lot of time on this software,is there a way to donate a few bucks?

Daniel

GrumpyOldPizza, 2019-03-08:
I suppose the best way to send me a thank you donation would be via:

https://www.jensonusa.com/Jenson-USA-Gift-Card

Thomas Roell, grumpyoldpizza@gmail.com

Guess I have always a need to upgrade piece of my road bikes ;-)

closed 2019-04-15 by GrumpyOldPizza