GrumpyOldPizza/ArduinoCore-stm32l0 #76: Arduino-LMIC

evandavey, 2019-03-25:
Is it possible to use this core with Arduino-LMIC rather than the LoRaWAN library?

GrumpyOldPizza, 2019-03-25:
No.

evandavey, 2019-03-25:
OK.  My device is getting stuck on joinOTAA for 2-3 minutes.  I have tested with another node running Arduino-LMIC (on different hardware) and the join is instant.  Both are next to the gateway and I can see good RSSI levels once the join has happened.

GrumpyOldPizza, 2019-03-25:
Just to clarify. The join takes time, but happens. 

What region are you using ? 

EU868 for example needs to respect a duty cycle requirement. So if the first join fails, it needs to wait some time.

What gateway are you using ?

evandavey, 2019-03-25:
Yes, join happens after time. I've been trying (so far unsuccessfully) to
find other code to test with to rule out hardware issues (it's a custom
board using the Murata Type ABZ module). Region is AU915. Gateway is
pi/RAK831 running the multi protocol forwarder. No issues with other node
running lmic which activates instantly and consistently.

GrumpyOldPizza, 2019-03-25:
Interesting.

What subband are you using ? Channel frequencies ?

AU915 is like US915, so speced for 64+8 channels. A typical gateway can only do 8+1 channels. So you need to do a LoRaWAN.setSubBand(subband). 1-8 selects the band, 0 reverts to a 64+8 setup.

I suspect that LMIC either by chance finds the proper subband (or you configured it) right away, or it does not respect the required backoff procedure.

evandavey, 2019-03-25:
I am using subband 2 on both the nodes and gateway I.e. matching TTN

evandavey, 2019-03-25:
I have just been doing some more testing.  I can see an initial burst of current when the device first resets (which looks like a tx attempt), then it just looks like it sits idles for 2-3 mins, then another burst where it activates.  It then uses SF9 for subsequent transmissions (the other node uses SF7).

evandavey, 2019-03-26:
I managed to get https://github.com/ARMmbed/mbed-os-example-lorawan running and joins are much quicker 4-5 seconds indicating the hardware is probably OK and there is something going on with the library.