GrumpyOldPizza/ArduinoCore-stm32l0 #77: Channel Free and Frequency selection

PAk-CatchFire, 2019-03-29:
Hi again.

I am checking yor code and found ibn the SX1276.c file a function that checks if the channel is free (SX1276IsChannelFree).
Could it be used to check the channel to avoid collisions?

On the other hand, does the frequency selected has to be well defined on a channel? Or could I use a non standard one like 867.45 for example?

Thank you

GrumpyOldPizza, 2019-04-15:
Yes, this function could be used for this very purpose. It's also exposed in FskRadio directly. I'd discourage this direct API use within the driver layer.

closed 2019-04-15 by GrumpyOldPizza

PAk-CatchFire, 2019-04-16:
Thank you. Do you have an example?
How long does it take to detect a collision?

GrumpyOldPizza, 2019-04-16:
https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/tree/master/libraries/FskRadio/examples/Fsk_Sense

Collision detection requirements and such are defined in your local FCC/ETSI equivalent regulations.