GrumpyOldPizza/ArduinoCore-stm32l0 #78: Serial skipping characters in Ubuntu

BoelD, 2019-04-07:
When using the Serial port on Ubuntu, it sometimes (and seemingly randomly) skips characters. I tried various cables, baudrates and tested with both the serial monitor and minicom and all gave the same effect. Uploading a sketch works flawless. In Windows, everything works like it should and no characters are skipped.

I looked around for similar problems, tried playing with XON/XOFF settings, but that didn't help.

I also experience problems opening the serial port. Then I get either of these two error messages:

`Board at /dev/ttyACM0 is not available`
`Error opening serial port '/dev/ttyACM0'. (Port busy)`

I made sure I do not have another serial monitor open already.

This could be a Linux problem. However, when I us a 'normal' Arduino (UNO), I can use the serial port without loosing characters. So there may be some link with the STM32L0 core.

Does anyone else have similar problems? Or even better: was able to solve this?

BoelD, 2019-04-07:
Digged a lilte bit further and found out that it may be a DTR / RTS related issue. When I initially try to open the serial monitor, I get the message that the port is busy. When I reset the device, I can open the monitor again. But I do miss characters in the output. When I enter `dmesg` in the console, I see this as the last messages:

```
[  651.167769] usb 3-3: USB disconnect, device number 16
[  651.167939] cdc_acm 3-3:1.0: failed to set dtr/rts
[  652.370690] usb 3-3: new full-speed USB device number 17 using xhci_hcd
[  652.520214] usb 3-3: New USB device found, idVendor=1209, idProduct=6665
[  652.520223] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  652.520227] usb 3-3: Product: Grasshopper
[  652.520232] usb 3-3: Manufacturer: Tlera Corporation
[  652.520236] usb 3-3: SerialNumber: 054737440000
[  652.521484] cdc_acm 3-3:1.0: ttyACM0: USB ACM device
```

The second line may tell a lot about the problem: failed to set DTR/RTS

GrumpyOldPizza, 2019-04-07:
Is this with low saturation or high saturation ? How many bytes per second are you trying to transmit ?

BoelD, 2019-04-07:
(Very) low saturation. I tried on 9600 baud and Serial.printed a simple counter and put a delay of 100ms between each print. Higher baudrates show the same pattern of (seemingly) random dropping one or more bytes.

GrumpyOldPizza, 2019-04-08:
Any pattern ? Like every <n> th character missing ?

BoelD, 2019-04-08:
It seemed pretty random. Sometimes just one character missing and sometimes
a lot of characters on a row.

A thought I have is that some other process is interfering with the serial
communication. That could also exain the 'port busy' warnings I am running
into. But so far, I haven't been able to find out.

I will do some more structured tests later this week.

BoelD, 2019-04-10:
After some searching I probably found the culprit. `modemmanager` is interfering with the USB ports. I removed it with:

`sudo apt-get purge modemmanager`

And so far I didn't have any problems anymore. The post busy and port not available messages are gone and all characters are displayed now.

GrumpyOldPizza, 2019-04-15:
Ok, this was in the README.md to install the relevant rules file for Linux ...

closed 2019-04-15 by GrumpyOldPizza