GrumpyOldPizza/ArduinoCore-stm32l0 #82: EEPROM code snipet

Taher1322, 2019-06-01:
Hello Sir, I want to store a particular value into eeprom and before transmitting have to read and transmit over the lora network. Can you help me out with basic read/write operation in Grasshopper board. Small code of device performing read/write operation would be helpful.

Thank you

GrumpyOldPizza, 2019-06-02:
The code is using the normal EEPROM class. 

https://www.arduino.cc/en/Reference/EEPROM

There are also examples included directly.

closed 2019-06-02 by GrumpyOldPizza

Taher1322, 2019-06-03:
Sir, I am aware of that Arduino reference. what i need to understand is how can i use EEPROM address locations on cmwx1zzabz and store a value and read it later. 
Hope to hear from you soon
Thank You

GrumpyOldPizza, 2019-06-03:
I am not sure what you are asking. The EEPROM class is the way to access
the EEPROM on STM32L082. There is code with the examples of how to do that.

If you are interested in how to directly access the underlying hardware,
then You might want to check out the reference manual.