GrumpyOldPizza/ArduinoCore-stm32l0 #84: TimerMillis() - does not work at some delay values

SloMusti, 2019-06-04:
Testing on B-L072Z-LRWAN1 with the latest release, I have observed that on some particular values, the timer does not work. Testing with provided example TimerMillis_Blink https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/master/libraries/TimerMillis/examples/TimerMillis_Blink/TimerMillis_Blink.ino#L35 changing the delay value from default 2000 to other values:

 * 8500 - does not work
 * 8000 - works fine
 * 9000- works fine

Have tested the values in other sketches, for example here: https://github.com/SloMusti/microclimate-lorawan-firmware/blob/master/src/main.ino#L45 and in certain instances it work work well with 8500 value as well. I highly doubt this issue is correlated directly to the timer values, however I have been able to repeatably reproduce this.

Note, this issues was present only during certain power-up instances on multiple boards, after power-cycling the complete board, either B-L072Z-LRWAN1 or Murata on a custom board, the problem does not persist. What scenario leads to this situation that Timer does not fire - reset the watchdog as typically used is unknown, however watchdog reset does not eliminate this issue, which is strange at best but also may lead to unpredictable product behavior.

If someone else can test and has ideas, it would be great to explore this.

SloMusti, 2019-06-08:
After extensive testing this odd issue appears to manifest due to having more TimerMillis objects in different parts of the sketch, reducing the number resolved the behavior. @GrumpyOldPizza can you please specify how many TimerMillis objects one can have and if you believe it is possible they do not get cleared upon system reset, only power cycle?

GrumpyOldPizza, 2019-06-10:
Do you have a simple repro sketch for the "too many timer objects" issue ?

SloMusti, 2019-06-23:
This issue is quite elusive, however I have manged to trace it back to initializing GNSS. This is only seen in the final application code, not for example GNSS examples, so it is quite unclear why that is: https://github.com/IRNAS/smartparks-rhino-tracker-firmware/tree/ca26bc8bf9d51bc99ca054c9b2100871c6a86595

From what I can see below, the timer fires two times earlier then required, then once on the schedule and never again until WDT reset. There may be cases when this works ok, but it will fail in the large majority of cases.

An example without GNSS initialized - same code overall
```
242251 -  setup(serial debug begin): resetCause: 1
525093 - fsm(0>0,-1,178)
525095 - fsm(0>1,-1,184)
525129 - fsm(1>2,-1,218)
525132 - fsm(2>3,-1,221)
525134 - sensor_init()
525136 - LIS present: 0x44
525138 - fsm(3>5,-1,226)
525158 - fsm(5>12,-1,248)
525169 - wdt(): 5177

530090 - fsm(12>12,-1,5178)
530092 - fsm(12>4,-1,5181)
530094 - fsm(4>6,-1,5184)
530127 - fsm(6>12,-1,5217)
530130 - wdt(): 10177

535090 - fsm(12>12,-1,10179)
535092 - fsm(12>4,-1,10182)
535095 - fsm(4>7,-1,10184)
535097 - fsm(7>10,-1,10187)
535100 - fsm(10>11,-1,10190)
535121 - fsm(11>12,-1,10211)
535124 - wdt(): 15177

540124 - fsm(12>12,-1,15214)
540127 - fsm(12>4,-1,15216)
545087 - wdt(): 20177
545089 - wdt(): 25177
```

An example with GNSS code initialized
```
580093 - setup(serial debug begin): resetCause: 1
596289 - fsm(0>0,-1,178)
596291 - fsm(0>1,-1,184)
596325 - fsm(1>2,-1,218)
596328 - fsm(2>3,-1,221)
596330 - sensor_init()
596332 - LIS present: 0x44
596334 - fsm(3>5,-1,226)
596354 - fsm(5>12,-1,248)
596368 - wdt(): 5177

601286 - fsm(12>12,-1,5178)
601288 - fsm(12>4,-1,5181)
601291 - fsm(4>6,-1,5184)
601323 - fsm(6>12,-1,5217)
601326 - wdt(): 10177

606286 - fsm(12>12,-1,10179)
606288 - fsm(12>4,-1,10182)
606291 - fsm(4>7,-1,10184)
606293 - gps(sensor_gps_init())
607550 - wdt(): 11443 ### This is ok

608683 - wdt(): 12576 ### This should not happen
608878 - wdt(): 12771 ### This should not happen

609600 - gps(initialized)
609603 - gps(started, timeout: 20000 , ehpe 40)
609607 - fsm(7>8,-1,13496)
609611 - fsm(8>8,-1,14502)
610611 - fsm(8>9,-1,14505)
611482 - wdt(): 16032 ### This is ok, but does not happen again

612481 - fsm(9>9,-1,18507)
614616 - fsm(9>9,-1,22510)
619488 - fsm(9>9,-1,26513)
623404 - gps(lat: 0.0000000 lon 0.0000000 alt: 0.000 hdop: 99.99 epe: 0.00 sat: 0 ttf: 15030.00)
624759 - fsm(9>9,-1,30516)
626624 - fsm(9>10,-1,30519)
626627 - fsm(10>11,-1,30521)
626648 - fsm(11>12,-1,30542)
626651 - setup(serial debug begin): resetCause: 3 ### This is WDT resetCause
```

SloMusti, 2019-07-05:
@GrumpyOldPizza  Further working on this to provide a minimal example to demonstrate how GNSS library messes up timers somehow.

Setup B-L072Z board with GPS connected on Serial1 and powered through two GPIO pins. Same example can be run without GPS connected, just commenting out the while busy line.

What happens is that once `GNSS.end()` is called, the periodic timer that is set-up fires randomly and then stops. Looking at the below printout shows this nicely:

```
2035 - gps_power(173)
2050 - wdt(1173)
3050 - wdt(2173)
4049 - wdt(3173)
5049 - wdt(4173)
6049 - wdt(5173)
7049 - wdt(6173)
8049 - wdt(7173)
9049 - wdt(8173)
10048 - wdt(9173)
11048 - gps_begin(10173)
12035 - wdt(10173)
12532 - wdt(10672)) // this should not happen until
12599 - wdt(10739)) // this should not happen until
13622 - wdt(11762)) // this should not happen until
13782 - wdt(11922) ) // this should not happen until
14252 - gps_end(12393)
14255 - sleep(12393) ) // this should should be at a different time to keep the 1s interval, note it happens next to GPS end
14268 - wdt(13001) #after this WDT should be called again but is not until 5s alter watchdog kick in
14867 - boot(173)
19999 - gps_power(173)
20006 - wdt(1173)
21006 - wdt(2173)
22006 - wdt(3173)
23005 - wdt(4173)
24005 - wdt(5173)
25005 - wdt(6173)
26004 - wdt(7173)
27004 - wdt(8173)
28003 - wdt(9173)
29003 - gps_begin(10172)
29999 - wdt(10173)
30527 - wdt(10703)
30561 - wdt(10737)
31592 - wdt(11768)
31714 - wdt(11890)
31848 - wdt(12024)
32120 - gps_end(12296)
32122 - sleep(12296)
```

Code example is attached.
[GNSS_wdt_gps_end_stop.zip](https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/files/3363705/GNSS_wdt_gps_end_stop.zip)

SloMusti, 2019-07-05:
An option may be that the core actually tops after GNSS>end() due to setting uart object to none here: https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/master/libraries/GNSS/src/GNSS.cpp#L386 Does not explain multiple timers fired.

GrumpyOldPizza, 2020-04-26:
Dropped some new RTC/TimerMillis code on github. Mind checking this got fixed now ?

SloMusti, 2020-04-27:
Have tried validating this, however the gps library and a few other things will need to be cleaned up before I can run this:

```/tmp/arduino_build_35499/sketch/GNSS_wdt_gps_end_stop.ino.cpp.o: In function `setup':
/home/musti/Downloads/GNSS_wdt_gps_end_stop/GNSS_wdt_gps_end_stop.ino:29: undefined reference to `TimerMillis::start(void (*)(), unsigned long, unsigned long)'
/tmp/arduino_build_35499/libraries/GNSS/utility/gnss_core.c.o: In function `gnss_location':
/home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/libraries/GNSS/src/utility/gnss_core.c:447: undefined reference to `stm32l0_rtc_set_adjust'
/home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/libraries/GNSS/src/utility/gnss_core.c:464: undefined reference to `stm32l0_rtc_set_calendar'
/tmp/arduino_build_35499/libraries/GNSS/utility/gnss_core.c.o: In function `ubx_wakeup':
/home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/libraries/GNSS/src/utility/gnss_core.c:3228: undefined reference to `stm32l0_rtc_get_calendar'
/tmp/arduino_build_35499/libraries/TimerMillis/TimerMillis.cpp.o: In function `TimerMillis::timeout(TimerMillis*, _stm32l0_rtc_timer_t*)':
/home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/libraries/TimerMillis/src/TimerMillis.cpp:159: undefined reference to `stm32l0_rtc_time_offset'
/home/musti/.arduino15/packages/TleraCorp/hardware/stm32l0/0.0.10/libraries/TimerMillis/src/TimerMillis.cpp:160: undefined reference to `stm32l0_rtc_time_subtract'`

GrumpyOldPizza, 2020-04-27:
Looks like you are picking up a mixture of sources from a 0.0.10 installation and files from github.

SloMusti, 2020-04-27:
You are correct, deployed the latest code and now it work.

Running the test from https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/issues/84#issuecomment-508832711 now results in correct behavior and no timers are triggered unexpectedly. This resolves this issue.

closed 2020-04-27 by SloMusti