GrumpyOldPizza/ArduinoCore-stm32l0 #85: Is input pin 10 compatible with DS18B20?

rmeldo, 2019-06-07:
It's probably my fault, but I thought I would ask anyway.

Is it possible to read a DS18B20 using pin 10?    I keep getting a -55C reading, whereas I should be getting ~20C.

I am using passive power with a 4k7 ohm pullup.

Thanks
Riccardo

GrumpyOldPizza, 2019-06-10:
I cannot see a reason why this would not be working with PA10 ... Does it work on other pins ?

GrumpyOldPizza, 2020-05-06:
Is this still an issue ?

matthijskooijman, 2020-12-02:
I would also think that any pin should work with the DS18B20, since that is OneWire which is typically bit-banged (unless you use some library that uses some other method and has specific requirements).

Lacking response from original poster, I would suggest closing this issue.