GrumpyOldPizza/ArduinoCore-stm32l0 #87: GNSS add UBX-NAV-AOPSTATUS (0x01 0x60) message on receive

SloMusti, 2019-06-08:
Receiving UBX-NAV-AOPSTATUS (0x01 0x60) message would be useful to determine the state of Autonomous engine - AssistNow Autonomous Status. Waiting for status to be 0, such that the ephemeris data is processed correctly before sleep, optimizing ttff over longer periods.

https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_%28UBX-13003221%29_Public.pdf page 314 for more details

SloMusti, 2019-06-09:
I have implemented the first version of the code to do this: https://github.com/SloMusti/ArduinoCore-stm32l0/commit/c134cc5d1a250b91b31d651c09138f2133a0e4bc

Currently it does not appear to work yet, given that aopcfg status is always shown as 0 as well as it does not appear UBX-NAV-AOPSTATUS  are being set at all, they may need to be enabled, however that is something yet to be figured out.

SloMusti, 2019-06-09:
Added also the message enable for aopstatus, however event this appears to be insufficient: https://github.com/SloMusti/ArduinoCore-stm32l0/commit/efe596965ac24374ee5b9dc024c6a0bbed993639

SloMusti, 2019-06-10:
Now verified that the GPS module is sending AOPSTATUS message.

```
08:39:20  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
08:39:20  R -> UBX NAV-SAT,  Size 256,  'Satellite Status and Information'
08:39:20  R -> UBX NAV-DOP,  Size  26,  'Dilution of Precision'
08:39:20  R -> UBX NAV-TIMEGPS,  Size  24,  'GPS System Time'
08:39:20  R -> UBX NAV-AOPSTATUS,  Size  24,  'AOP Status'

08:39:20  0000  B5 62 01 07 5C 00 10 19 02 07 E3 07 06 0A 08 27  µb..\.....ã....'
          0010  14 37 60 92 0B 05 90 B9 F8 FF 00 00 24 00 43 4D  .7`....¹øÿ..$.CM
          0020  55 09 C5 47 C1 1B 92 26 05 00 79 82 04 00 80 DD  U.ÅGÁ..&..y....Ý
          0030  63 52 C0 33 42 3A 00 00 00 00 00 00 00 00 00 00  cRÀ3B:..........
          0040  00 00 00 00 00 00 00 00 00 00 20 4E 00 00 80 A8  .......... N...¨
          0050  12 01 0F 27 00 00 46 B8 23 23 00 00 00 00 00 00  ...'..F¸##......
          0060  00 00 A6 26                                      ..¦&.
          
08:39:20  0000  B5 62 01 35 F8 00 10 19 02 07 01 14 00 00 00 02  µb.5ø...........
          0010  0D 27 67 00 00 00 13 5C 00 00 00 06 12 1B 39 00  .'g....\......9.
          0020  00 00 13 5C 00 00 00 0B 00 C9 60 01 00 00 11 12  ...\.....É`.....
          0030  00 00 00 0C 00 46 30 00 00 00 11 5C 00 00 00 0E  .....F0....\....
          0040  00 1A 2F 01 00 00 11 5C 00 00 00 13 0A 09 2D 00  ../....\......-.
          0050  00 00 13 5C 00 00 00 1D 00 23 D7 00 00 00 11 12  ...\.....#×.....
          0060  00 00 00 1F 00 0B 38 01 00 00 11 5C 00 00 00 20  ......8....\... 
          0070  00 1F 19 01 00 00 11 5C 00 00 01 78 00 1C DC 00  .......\...x..Ü.
          0080  00 00 01 07 00 00 01 7B 00 22 9F 00 00 00 01 07  .......{."......
          0090  00 00 01 88 00 23 C3 00 00 00 01 07 00 00 06 07  .....#Ã.........
          00A0  00 0E 1F 01 00 00 11 1A 00 00 06 08 00 0C 4F 01  ..............O.
          00B0  00 00 11 12 00 00 06 09 00 1D 37 01 00 00 11 12  ..........7.....
          00C0  00 00 06 0F 00 1B 6F 00 00 00 11 1A 00 00 06 10  ......o.........
          00D0  00 4A 32 00 00 00 11 1A 00 00 06 11 00 1F 23 00  .J2...........#.
          00E0  00 00 11 12 00 00 06 12 00 46 68 00 00 00 11 12  .........Fh.....
          00F0  00 00 06 13 00 22 B6 00 00 00 11 12 00 00 A9 8C  ....."¶.......©..
          
08:39:20  0000  B5 62 01 04 12 00 10 19 02 07 0F 27 0F 27 0F 27  µb.........'.'.'
          0010  0F 27 0F 27 0F 27 0F 27 C3 E3                    .'.'.'.'Ãã.
          
08:39:20  0000  B5 62 01 20 10 00 10 19 02 07 90 B9 F8 FF 09 08  µb. .......¹øÿ..
          0010  12 07 60 92 0B 05 CF 3C                          ..`...Ï<.
          
08:39:20  0000  B5 62 01 60 10 00 10 19 02 07 01 00 00 0A 00 00  µb.`............
          0010  00 00 00 00 00 00 AE A8                          ......®¨.
```

GrumpyOldPizza, 2019-06-10:
I suppose I could add that. My concern is more the added data over USART, which costs extra power. Also only aopStatus() is needed. Might do this in the future along with spoofing detection.

SloMusti, 2019-06-10:
@GrumpyOldPizza this message is only enabled if `Autonomous mode is enabled`, please see https://github.com/SloMusti/ArduinoCore-stm32l0/commit/efe596965ac24374ee5b9dc024c6a0bbed993639 Given I have verified that GPS is actually sending this and the code still does not get the flags sent form the messages, I must have missed something in implementation. Would be great if you could point me in the right direction.

GrumpyOldPizza, 2019-06-10:
My point was not that if you manually enable Autonomous mode, why would you need to waste cycles to add the flag to the GNSSLocation ? What you want is whether AOP is done computing before you turn off the GNSS.

SloMusti, 2019-06-10:
@GrumpyOldPizza as far as I understand the GPS will be sending the AOP status data bit, however it will be changing from 0 to 1 and back during the operation and one should not turn off while it is set to 1. Upon boot/initially the signal does not go to 1 and stay there until completed, but only intermittently jumps to 1.

Unfortunately it does not seem to be the case that the receiver would simply change the value once done. 

Agreed that is could be implemented differently then using a flag, your suggestion would be appreciated what would be a better method.

GrumpyOldPizza, 2020-05-02:
Looks to me like this is only needed for the suspend/resume logic. I.e. poll the message till the engine is idle ?

SloMusti, 2020-05-02:
Yes, this means that you delay a bit going to suspend. Havent gotten very far in testing this however.