GrumpyOldPizza/ArduinoCore-stm32l0 #89: LoRaWAN Stack Documentation

hallard, 2019-06-22:
Hi,
More a question than an issue. 
I was not able to find documentation on the LoRaWAN stack implementation of this repo (mainly `LoRaWAN.cpp` file) so my question is, is there any somewhere about parameters values passed to function and so?
I looked into the code and examples and looks very fine, also saving all in EEP. Did you write it from scratch or are you using high level function over the default semtech LoRaMac-node one?
Thanks for your help

SloMusti, 2019-06-22:
@hallard We are using this library successfully for LoraWAN, although there is no documentation per-say, examples have been generally sufficient to understand what is possible and implement all the features.

hallard, 2019-06-23:
@SloMusti I have no doubts it works perfectly, my thought was more to be sure I'm using it correctly