GrumpyOldPizza/ArduinoCore-stm32l0 #91: EEPROM addresses: which ones are reserved?

BoelD, 2019-07-06:
I want to use the EEPROM to store some data and need to know which EEPROM addresses are reserved for the commissioning data. I had a look in LoRaWAN.cpp, but that didn't help me.

To rephrase the question: which address range can I use with the EEPROM class, without overwriting the commissoning data stored in the same EEPROM?

SloMusti, 2019-07-11:
I have managed to save data as shown here:

https://github.com/IRNAS/smartparks-rhino-tracker-firmware/blob/development_gps_tracker/src/settings.h#L9
https://github.com/IRNAS/smartparks-rhino-tracker-firmware/blob/development_gps_tracker/src/settings.cpp#L44

The address I used without overwriting everything is 0 as the libraries use some offset.

hallard, 2019-07-11:
looks like It's on the LoRaWAN.cpp file

https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0/blob/master/libraries/LoRaWAN/src/LoRaWAN.cpp#L47-L68

```cpp
#if defined(DATA_EEPROM_BANK2_END)
#define EEPROM_OFFSET_START            ((((DATA_EEPROM_BANK2_END - DATA_EEPROM_BASE) + 1023) & ~1023) - 1024)
#else
#define EEPROM_OFFSET_START            ((((DATA_EEPROM_END - DATA_EEPROM_BASE) + 1023) & ~1023) - 1024)
#endif

#define EEPROM_OFFSET_COMMISSIONING    (EEPROM_OFFSET_START + 0)
#define EEPROM_OFFSET_SESSION          (EEPROM_OFFSET_START + 128)
#define EEPROM_OFFSET_PARAMS           (EEPROM_OFFSET_START + 256)
#define EEPROM_OFFSET_DEVNONCE         (EEPROM_OFFSET_START + 512)
#define EEPROM_OFFSET_UPLINK_COUNTER   (EEPROM_OFFSET_START + 512 + 128)
#define EEPROM_OFFSET_DOWNLINK_COUNTER (EEPROM_OFFSET_START + 512 + 256)
#define EEPROM_OFFSET_RESERVED         (EEPROM_OFFSET_START + 512 + 384)

#define EEPROM_SIZE_COMMISSIONING      128
#define EEPROM_SIZE_SESSION            128
#define EEPROM_SIZE_PARAMS             256
#define EEPROM_SIZE_DEVNONCE           128
#define EEPROM_SIZE_UPLINK_COUNTER     128
#define EEPROM_SIZE_DOWNLINK_COUNTER   128
#define EEPROM_SIZE_RESERVED           128
```

So it start from 0x1400 and goes to 0x17FF

use that sketch that will display what you need
```cpp
/*
 * EEPROM Read
 *
 * Reads the value of each byte of the EEPROM and prints it
 * to the computer.
 * This example code is in the public domain.
 */

#include <LoRaWAN.h>
#include <EEPROM.h>

#if defined(DATA_EEPROM_BANK2_END)
#define EEPROM_OFFSET_START            ((((DATA_EEPROM_BANK2_END - DATA_EEPROM_BASE) + 1023) & ~1023) - 1024)
#else
#define EEPROM_OFFSET_START            ((((DATA_EEPROM_END - DATA_EEPROM_BASE) + 1023) & ~1023) - 1024)
#endif

#define EEPROM_OFFSET_COMMISSIONING    (EEPROM_OFFSET_START + 0)
#define EEPROM_OFFSET_SESSION          (EEPROM_OFFSET_START + 128)
#define EEPROM_OFFSET_PARAMS           (EEPROM_OFFSET_START + 256)
#define EEPROM_OFFSET_DEVNONCE         (EEPROM_OFFSET_START + 512)
#define EEPROM_OFFSET_UPLINK_COUNTER   (EEPROM_OFFSET_START + 512 + 128)
#define EEPROM_OFFSET_DOWNLINK_COUNTER (EEPROM_OFFSET_START + 512 + 256)
#define EEPROM_OFFSET_RESERVED         (EEPROM_OFFSET_START + 512 + 384)

#define EEPROM_SIZE_COMMISSIONING      128
#define EEPROM_SIZE_SESSION            128
#define EEPROM_SIZE_PARAMS             256
#define EEPROM_SIZE_DEVNONCE           128
#define EEPROM_SIZE_UPLINK_COUNTER     128
#define EEPROM_SIZE_DOWNLINK_COUNTER   128
#define EEPROM_SIZE_RESERVED           128

#define EEPROM_TOTAL_SIZE EEPROM_SIZE_COMMISSIONING + \
                          EEPROM_SIZE_SESSION + \
                          EEPROM_SIZE_PARAMS + \
                          EEPROM_SIZE_DEVNONCE + \
                          EEPROM_SIZE_UPLINK_COUNTER + \
                          EEPROM_SIZE_DOWNLINK_COUNTER + \
                          EEPROM_SIZE_RESERVED 

void eep_hex_dump(const char * name, int addr, int len){
  int i;
  char buff[17];
  char str[64];
  byte value;

  sprintf(str, "eep_hex_dump %s size:%3d bytes starting at 0x%04X", name, len, addr);
  Serial.println(str);
  Serial.println();
}

void setup() {
  // initialize serial and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.print("EEPROM_OFFSET_START : 0x");
  Serial.println(EEPROM_OFFSET_START, HEX);

  eep_hex_dump("EEPROM_OFFSET_COMMISSIONING", EEPROM_OFFSET_COMMISSIONING, EEPROM_SIZE_COMMISSIONING);
  eep_hex_dump("EEPROM_OFFSET_SESSION", EEPROM_OFFSET_SESSION, EEPROM_SIZE_SESSION);
  eep_hex_dump("EEPROM_OFFSET_PARAMS", EEPROM_OFFSET_PARAMS, EEPROM_SIZE_PARAMS);
  eep_hex_dump("EEPROM_OFFSET_DEVNONCE", EEPROM_OFFSET_DEVNONCE, EEPROM_SIZE_DEVNONCE);
  eep_hex_dump("EEPROM_OFFSET_UPLINK_COUNTER", EEPROM_OFFSET_UPLINK_COUNTER, EEPROM_SIZE_UPLINK_COUNTER);
  eep_hex_dump("EEPROM_OFFSET_DOWNLINK_COUNTER", EEPROM_OFFSET_DOWNLINK_COUNTER, EEPROM_SIZE_DOWNLINK_COUNTER);
  eep_hex_dump("EEPROM_OFFSET_RESERVED", EEPROM_OFFSET_RESERVED, EEPROM_SIZE_RESERVED);

  Serial.print("EEPROM_OFFSET_END   : 0x");
  Serial.println(EEPROM_OFFSET_START + EEPROM_TOTAL_SIZE, HEX);

}

void loop() {
  delay(5000);
}
```

and display something like that on a fresh chip
```
EEPROM_OFFSET_START : 0x1400
eep_hex_dump EEPROM_OFFSET_COMMISSIONING size:128 bytes starting at 0x1400
eep_hex_dump EEPROM_OFFSET_SESSION size:128 bytes starting at 0x1480
eep_hex_dump EEPROM_OFFSET_PARAMS size:256 bytes starting at 0x1500
eep_hex_dump EEPROM_OFFSET_DEVNONCE size:128 bytes starting at 0x1600
eep_hex_dump EEPROM_OFFSET_UPLINK_COUNTER size:128 bytes starting at 0x1680
eep_hex_dump EEPROM_OFFSET_DOWNLINK_COUNTER size:128 bytes starting at 0x1700
eep_hex_dump EEPROM_OFFSET_RESERVED size:128 bytes starting at 0x1780
EEPROM_OFFSET_END   : 0x1800

```

matthijskooijman, 2020-12-02:
AFAIU, you can use "all" of EEPROM through the `EEPROM` class, since that class (and the `E2END` macro) are artificially limited to report a smaller size for the eeprom, automatically reserving the upper 2K or so.

See also #166 for some related discussion.