GrumpyOldPizza/ArduinoCore-stm32l0 #92: compilation issue when creating custom board out of the package

hallard, 2019-07-08:
Hi There,
I usually create custom boards for Arduino IDE that I place in `hardware\` folder and then I can select these board from IDE and compile + upload. In this case I never need to download to my hardware folder all compile tools and other.
See example I made for some LoRa boards [there](https://github.com/ch2i/ch2i-arduino-boards)

Note that I installed this repo with board manager, I can see and upload to grasshopper board without any issue.

I tried to create a new `hardware\stm32l0\` folder where I put my custom murata board definition (based on staff you done for this repo)

I can see my new board and select it, but the issue occurs on compile time, looks like it can't see the framework location (IE include files, library, ...). 
As this technique works for official Arduino board (328p or SAM21) I'm wondering what I'm doing wrong there ?

Here below the the folder tree I'm using.
[image]

How can I tell (maybe in boards.txt or platform.txt) to use the already installed (with board manager) ArduinoCore-stm32l0 from here for compilation?

I tried all of these, but no success
```
CH2i-Murata-L082CZ.build.core=arduino:stm32l0
CH2i-Murata-L082CZ.build.core=stm32l0
CH2i-Murata-L082CZ.build.core=arduino:TleraCorp
CH2i-Murata-L082CZ.build.core=TleraCorp
CH2i-Murata-L082CZ.build.core=arduino
```

An if I use the default
```
CH2i-Murata-L082CZ.build.core=arduino
```

same error
```

S:\PortableApps\arduino\arduino-builder -dump-prefs -logger=machine -hardware S:\PortableApps\arduino\hardware -hardware C:\Users\Charles\AppData\Local\Arduino15\packages -hardware S:\Skydrive\devt\Arduino\hardware -tools S:\PortableApps\arduino\tools-builder -tools S:\PortableApps\arduino\hardware\tools\avr -tools C:\Users\Charles\AppData\Local\Arduino15\packages -built-in-libraries S:\PortableApps\arduino\libraries -libraries S:\Skydrive\devt\Arduino\libraries -fqbn=CH2i:stm32l0:CH2i-Murata-L082CZ:usb=cdc,speed=32,opt=os -vid-pid=0403_6001 -ide-version=10809 -build-path C:\Users\Charles\AppData\Local\Temp\arduino_build_799357 -warnings=default -build-cache C:\Users\Charles\AppData\Local\Temp\arduino_cache_161664 -prefs=build.warn_data_percentage=75 -verbose S:\Skydrive\devt\STM32\CMWX1ZZABZ\CH2i_Murata\Blink_RGB\Blink_RGB.ino
S:\PortableApps\arduino\arduino-builder -compile -logger=machine -hardware S:\PortableApps\arduino\hardware -hardware C:\Users\Charles\AppData\Local\Arduino15\packages -hardware S:\Skydrive\devt\Arduino\hardware -tools S:\PortableApps\arduino\tools-builder -tools S:\PortableApps\arduino\hardware\tools\avr -tools C:\Users\Charles\AppData\Local\Arduino15\packages -built-in-libraries S:\PortableApps\arduino\libraries -libraries S:\Skydrive\devt\Arduino\libraries -fqbn=CH2i:stm32l0:CH2i-Murata-L082CZ:usb=cdc,speed=32,opt=os -vid-pid=0403_6001 -ide-version=10809 -build-path C:\Users\Charles\AppData\Local\Temp\arduino_build_799357 -warnings=default -build-cache C:\Users\Charles\AppData\Local\Temp\arduino_cache_161664 -prefs=build.warn_data_percentage=75 -verbose S:\Skydrive\devt\STM32\CMWX1ZZABZ\CH2i_Murata\Blink_RGB\Blink_RGB.ino
Using board 'CH2i-Murata-L082CZ' from platform in folder: S:\Skydrive\devt\Arduino\hardware\CH2i\stm32l0
Using core 'arduino' from platform in folder: S:\Skydrive\devt\Arduino\hardware\CH2i\stm32l0
Detecting libraries used...
"C:\\Users\\Charles\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions -w -x c++ -E -CC -D_SYSTEM_CORE_CLOCK_=32000000L -DARDUINO=10809 -DARDUINO_STM32L0_CH2I_MURATA -DARDUINO_ARCH_STM32L0 -DSTM32L082xx -march=armv6-m -mthumb -mabi=aapcs -mfloat-abi=soft -fsingle-precision-constant -DUSB_VID=0x1209 -DUSB_PID=0x6666 -DUSB_DID=0xffff "-DUSB_MANUFACTURER=\"CH2i\"" "-DUSB_PRODUCT=\"CH2i Murata L082CZ\"" -DUSB_TYPE=USB_TYPE_CDC "-IS:\\Skydrive\\devt\\Arduino\\hardware\\CH2i\\stm32l0/system/CMSIS/Include" "-IS:\\Skydrive\\devt\\Arduino\\hardware\\CH2i\\stm32l0/system/CMSIS/Device/ST/STM32L0xx/Include" "-IS:\\Skydrive\\devt\\Arduino\\hardware\\CH2i\\stm32l0/system/STM32L0xx/Include" "-IS:\\Skydrive\\devt\\Arduino\\hardware\\CH2i\\stm32l0\\cores\\arduino" "-IS:\\Skydrive\\devt\\Arduino\\hardware\\CH2i\\stm32l0\\variants\\CH2i-Murata-L082CZ" "C:\\Users\\Charles\\AppData\\Local\\Temp\\arduino_build_799357\\sketch\\Blink_RGB.ino.cpp" -o nul
C:\Users\Charles\AppData\Local\Temp\arduino_build_799357\sketch\Blink_RGB.ino.cpp:1:21: fatal error: Arduino.h: No such file or directory

 #include <Arduino.h>

                     ^

compilation terminated.

exit status 1
Error compiling for board CH2i Murata L082CZ.

```
Any idea of what I could do?
And I really would like to avoid a fork and add a sync between repos, I already have too much of this kind

Oh of course If I put my modification for my custom board in `boards.txt` of this repo (so under `Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.10` on my computer) it works, but this will be overwrite on your next release update.

SloMusti, 2019-07-11:
+1 on figuring this out as I also have a number of custom boards and would be good to be able to have thing synced and available

GrumpyOldPizza, 2020-05-06:
There is a documented way to do that (if you actually can find it). Need to put a sample on github, so that other folks can do their own boards.

nelisse, 2020-05-06:
+1 I was trying to do exactly the same withou much success, an example/description would be very useful!
For now I added my boards to the boards.txt file in C:\Users\<USER>\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32l0\0.0.10 and added a folder for each within 'variants' (also added driver defintions to both linux and windows in 'drivers' folder)

nelisse, 2020-05-08:
Hallard & SlowMusti: 
I also developed some Murata based boards and like to do the same, my own 'package' next to the STM32L0 package by GrumpyOldPizza using the tools from his package.
To build my code I created a Make setup using the Arduino CLI
This now compiles all my code for all boards (incl TleraCorp, STM Nucleo, STM Discovery) I use.
Within the Arduino CLI there is some decent info, including on the 'Platform specification', see (https://arduino.github.io/arduino-cli/platform-specification/) 
However I still cannot find exactly what I am looking for.
Did you have some success in the mean time?

nelisse, 2020-05-08:
I forgot the mention:
and the 'package_index.json specification' (see https://arduino.github.io/arduino-cli/package_index_json-specification/)